SpiNNakerManchester/SpiNNFrontEndCommon #11: Make install doesn't copy Makefile.SpiNNFrontEndCommon if it's already present

AlexRast, 2015-07-07:
If you run make clean, then make all, or make install, and a Makefile.SpiNNFrontEndCommon already exists in your $(SPINN_DIRS)/make_lib directory, the make won't overwrite the old version. Thus if you have an older install that you're upgrading, following the upgrade, compiles are likely to fail.

My suggested solution is to remove the file in $(SPINN_DIRS)/make_lib if you run make clean for SpiNNFrontEndCommon.

rowleya, 2015-07-07:
If, when in SpiNNFrontEndCommon/c_common I run:
>touch front_end_common_lib/Makefile.SpiNNFrontEndCommon
>make

I see it clearly copying the Makefile into place:

>for d in front_end_common_lib; do (cd $d; "make" install) || exit $?; done
>make[1]: Entering directory `/c/Users/zzalsar4/git/SpiNNFrontEndCommon/c_common/front_end_common_lib'
>cp Makefile.SpiNNFrontEndCommon /c/Users/zzalsar4/git/spinnaker_tools/make_lib/Makefile.SpiNNFrontEndCommon
>make[1]: Leaving directory `/c/Users/zzalsar4/git/SpiNNFrontEndCommon/c_common/front_end_common_lib'
...

I can only assume that you somehow had a newer version in the make_lib folder.

It is possible, from within SpiNNFrontEndCommon/c_common/front_end_common_lib, to run:
>make install-clean

This will remove everything that SpiNNFrontEndCommon installs in $(SPINN_DIRS)

AlexRast, 2015-07-08:
> If, when in SpiNNFrontEndCommon/c_common I run:
>
> I see it clearly copying the Makefile into place:
>
> `/c/Users/zzalsar4/git/SpiNNFrontEndCommon/c_common/front_end_common_lib' [...]

I looked at the versions and determined the following:

The version that had been in the make_lib folder originally was definitively 
older than the new one downloaded from the latest git master, *according to the 
authoritative copy of it held in the SpiNNFrontEndCommon source*. *That* version 
was from Sam's reload and was dated 22 May. The one in the git master is dated 1 
July.

*But*, when you run a make, and it copies it over into make_lib, it dates the 
copy according to the *copy* time, not according to the original modification 
time of the file. In this case the install with Sam's reload version was done on 
1 July, as it happens at a later time of the day than the newest makefile was 
created. So it apparently would not make the copy in this case.

I think this is the wrong behaviour. The copy shouldn't care when the last 
version was copied across, relative to the modification time of the new version 
to be copied - it should care when the modification time of the last version 
actually is, relative to the modification time of the new version. Otherwise 
someone who runs an install with an "older" version of the toolchain, then 
attempts to run an install with a "newer" version whose modification time is 
earlier than the last time the user tried to run the install, will have various 
things possibly out-of-date. Either the make should change the modification time 
on the newly copied file to match that of its corresponding original, or it 
should flatly overwrite the file, regardless of whether the version it sees is 
older (a not unreasonable thing to do, given that our toolchain for the moment 
remains "tightly coupled" to specific versions), or it should absolutely delete 
existing make targets, during make clean.

rowleya, 2015-07-08:
I just did a test – the time on the file that has been pulled from the repository or cloned is the time at which this is done, rather than the time that the file was written to the repository.  So if you revert to an older version and then make install, it should copy this version as well, as you will have updated the file.  To be clear, the file will not have a new date from a git pull if the file itself hasn’t changed, but if it has, it will get the timestamp at which you check it out; no timestamp modification is done by git or make by default.  I believe this is the correct mechanism, and I can say that I have never had this problem; in any case, we have no control over this mechanism, as it is the built in behavior of make and git.

It looks like what might have happened in your case is that you have two copies of the repository (or at the very least two copies of the software).  In this case, it is essentially random as to the timestamps in each, as it depends on when you checked out the software, not when you ran “make install”.  In this case, I would suggest that you keep two entirely separate environments, so that you don’t have crossover between files in this way.  Unless you do this, you will have to be much more careful about what version is in place at any one time.

alan-stokes, 2015-07-20:
Howdi all, 

so im not sure how useful this is, but it seemed to help me with compilation issues. Its a script with assuems all your modules of the tool chain are installed within the same directory (as is with mine), it follows the steps needed to compile all the modules correctly, with the added benfit of cleaning all these libuary stuff as it goes. 

the contents of the bash script are shown below, so copy and paste into a sh file and execute, and it may just stop this issue, as it could be that your forgetting one line/ getting it in the wrong order in the list of actual commands you need to execute.

cd spinnaker_tools
source setup
cd ..
cd spinn_common
make clean
make || exit $?
make install
cd ..
cd SpiNNFrontEndCommon/c_common/
cd front_end_common_lib/
make install-clean
cd ..
make clean
make || exit $?
make install
cd ../..
cd sPyNNaker/neural_modelling/
make clean
make || exit $?
source setup
cd ../../SpiNNakerGraphFrontEnd/examples/
make clean
make || exit $?
#cd ../../sPyNNakerExtraModelsPlugin/neural_modelling
#make clean
#make || exit $?

hope this helps you.

closed 2015-08-13 by alan-stokes