SpiNNakerManchester/SpiNNFrontEndCommon #73: Buffering-In Interface

rowleya, 2016-09-27:
The buffering out of packets is quite general but buffering in is restricted to working with multicast packets using EIEIO using the ReverseIPTagMulticastSource.  This could be generalised by having a more general wrapping protocol similar to the current EIEIO commands that then doesn't care about what is in the space being buffered in, just that some space is available for more data.

alan-stokes, 2016-09-28:
on top of this. the buffered out is an actual interface which allows end users to use it via the recording stuff. Buffered in on the other hand is limited to the impl in RIPMCS. During this refactor (or complimentary to this) the buffered in should become a simular interface, allowing other models/c code to have buffered in data regions.

dkfellows, 2021-03-26:
Any current use case for doing this?

rowleya, 2021-03-26:
Moved to bluesky again, as no specific pressing need.