SpiNNakerManchester/SpiNNFrontEndCommon #75: Support buffering of IO_BUF

rowleya, 2016-09-28:
It is common for IO_BUF to become full when debugging a large network.  It would be useful therefore to support the pausing of the simulation and the extraction of the buffers during simulation.

Issues to solve include knowing how long to run the simulation for before pausing; one thought is that since this mode is debug, it could be very short runs between extraction e.g. of the order 100 time steps.  This could be computed if we know how much memory will be used over a given number of time steps but this could be hard for IO_BUF.

dkfellows, 2021-03-26:
This isn't as pressing any more now that we use a more compact in-memory representation of logging statements. Is it worth the effort to work on this any more?

rowleya, 2021-03-26:
I have moved it to bluesky.  Simply that it could be done if we run out of things to do otherwise!

dkfellows, 2023-09-27:
I'm guessing we're just not going to do this. The log dictionary work was sufficient after all.

closed 2023-09-27 by dkfellows