SpiNNakerManchester/SpiNNFrontEndCommon #79: Config system

neworderofjamie, 2016-10-20:
Frontend common requires a config file, containing a very specific set of keys but the code to create this is duplicated in sPyNNaker and the GFE. Should it perhaps be moved into frontendcommon?

alan-stokes, 2016-10-20:
Hi Jamie,

the move for that was 2 fold.
1. It allows a front end that doesn't want to use the config object, to
still be usable via the FEC as long as they could provide the same
functionality via their own object (aka  as long as they support the
interface of the config).
2. The config file is different for each front end, and so we cant assume
they use the same file (but mostly due to 1).

my recommendation would be for you to have a spynnaker_8.cfg file and run
with the interface, rather than removing this flexibility.

regards
Alan

neworderofjamie, 2016-10-20:
Also, the naming convention of the config keys is very inconsistent e.g.
``max_application_binaries_kept``, ``defaultapplicationdatafilepath`` and
``defaultReportFilePath``. Also even if you aren't using the data spec
system frontend common insists on you having a SpecExecution section

alan-stokes, 2016-10-20:
Having discussed with jamie and rowley, we've come to a compromise which we
think meets all constraints.

instead of bringing the config file down to fec, we can make the __init__
of the spinnaker_main_interface to expand its inputs to all the parameters
which it uses from the config. This is safe (we believe) as we only ever
use it for reading within that class, it never moves away from there.

This means the FE's need to expand from whatever config interface they use
to these parameter sets before going down to the FEC interface, and
therefore supports FE's which don't want to use a config parser, and allows
the current FE's to not need to change significantly.

On regards to PyNN 0.8 FE, we still want to use the config file format, for
consistency with previous end users who are using pynn 0.75 and thus
backward compatibility of previous releases etc.

 regards
Alan

alan-stokes, 2016-10-20:
on the format of keys, i give that to you 100%, ive tried to keep them
consistent with python interface, but it seems different developers have
wanted camal case etc. Don't know why the "defaultapplicationdatafilepath"
exists as that, as that follows no convention. Feel free to fix in all FE's
if you want, but we'll need to warn end users in the next release of the
change to support backwards compatibility.

"Also even if you aren't using the data spec system frontend common insists
on you having a SpecExecution section"  given the new inputs branch, this
could be fixed, but until then, alas everything has been using a DSE and so
that flag "ExecDSEOnHostFlag" has been essential. What we can do is expose
another 2 parameters at the init level which is optional_load_algoritms and
required_load_algorithms and use them instead, it'll push the Spynnaker and
GFE to do this logic above to build those lists, but thats not too bad.
Would that solve it for you?

regards
Alan

neworderofjamie, 2016-10-20:
Sounds good

alan-stokes, 2016-10-20:
Can you put that bit as a separate issue then. Thanks
Alan

Christian-B, 2017-07-27:
In the end the config file was moved down to front end common,

closed 2017-07-27 by Christian-B