SpiNNakerManchester/SpiNNFrontEndCommon #142: multiple runs should be plonked into their own run folders in reports

alan-stokes, 2017-06-13:
seems this functionality is bust, and instead its plonking new instances of report files inside run1 instead.......

rowleya, 2017-06-13:
Looks like the check is for has_ran and application_graph_changed and has_reset_last:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/interface/abstract_spinnaker_base.py#L612-L614

alan-stokes, 2017-06-13:
makes me wonder what our semantics of the run folder should be, new changes or just run?

alan-stokes, 2017-07-28:
well t moment its new folder per reset. which overall is sound. guess it would need a rename of run, but theres nothing better that comes to mind. so meh. Also pynn 0.8 has the same level of behaviour with reset and neo segments, so keeping it consistent is a bonus. thus closing issue

closed 2017-07-28 by alan-stokes