SpiNNakerManchester/SpiNNFrontEndCommon #149: remote_spinnaker_url causes weird bug

Christian-B, 2017-06-26:
I accidentally set remote_spinnaker_url when I intended to set machineName

However this produced an weird exception trace with Rolwley says shows a bug.

Traceback (most recent call last):
  File "/brenninc/spinnaker/sPyNNaker8/p8_integration_tests/tests_for_param_vertex_retrieval_from_board_after_1_run/test_synfire_20n_20pc_delays_delay_extensions_all_recording.py", line 35, in test_all_no_constarint
    record_gsyn_inh=False)
  File "/brenninc/spinnaker/sPyNNaker8/p8_integration_tests/scripts/synfire_run.py", line 371, in do_run
    p.run(run_times[-1])
  File "/brenninc/spinnaker/sPyNNaker8/spynnaker8/__init__.py", line 476, in run
    return __pynn_run(simtime, callbacks=callbacks)
  File "/home/M04/acs/brenninc/pynn8/lib/python2.7/site-packages/pyNN/common/control.py", line 110, in run
    return run_until(simulator.state.t + simtime, callbacks)
  File "/home/M04/acs/brenninc/pynn8/lib/python2.7/site-packages/pyNN/common/control.py", line 92, in run_until
    simulator.state.run_until(time_point)
  File "/brenninc/spinnaker/sPyNNaker8/spynnaker8/spinnaker.py", line 123, in run_until
    self._run(tstop - self._current_run_timesteps)
  File "/brenninc/spinnaker/sPyNNaker8/spynnaker8/spinnaker.py", line 168, in _run
    AbstractSpiNNakerCommon.run(self, duration_ms)
  File "/brenninc/spinnaker/sPyNNaker/spynnaker/pyNN/abstract_spinnaker_common.py", line 326, in run
    AbstractSpinnakerBase.run(self, run_time)
  File "/brenninc/spinnaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/abstract_spinnaker_base.py", line 652, in run
    self._get_machine(total_run_time, n_machine_time_steps)
  File "/brenninc/spinnaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/abstract_spinnaker_base.py", line 1067, in _get_machine
    inputs, algorithms, outputs)
  File "/brenninc/spinnaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/abstract_spinnaker_base.py", line 882, in _run_algorithms
    do_timings=self._do_timings, print_timings=self._print_timings)
  File "/brenninc/spinnaker/PACMAN/pacman/executor/pacman_algorithm_executor.py", line 127, in __init__
    use_unscanned_annotated_algorithms)
  File "/brenninc/spinnaker/PACMAN/pacman/executor/pacman_algorithm_executor.py", line 207, in _set_up_pacman_algorithm_listings
    optional_algorithms_datas, converter_algorithms_datas)
  File "/brenninc/spinnaker/PACMAN/pacman/executor/pacman_algorithm_executor.py", line 306, in _determine_algorithm_order
    algorithm_input_requirement_breakdown))
PacmanConfigurationException: Unable to deduce a future algorithm to use.
    Inputs: set(['BootPortNum', 'MachineAllocationController', 'TotalRunTime', 'ResetMachineOnStartupFlag', 'CPUsPerVirtualChip', 'MemoryApplicationGraph', 'MemoryExtendedMachine', 'TotalMachineTimeSteps', 'ScampConnectionData', 'MaxSDRAMSize', 'MachineWidth', 'MachineHeight', 'IPAddress', 'MemoryMachineGraph', 'NChipsRequired', 'DownedChipsDetails', 'RemoteSpinnakerUrl', 'BMPDetails', 'BoardVersion', 'MemoryMachine', 'DownedCoresDetails', 'MemoryGraphMapper', 'EnableReinjectionFlag', 'MachineHasWrapAroundsFlag', 'MachineTimeStep', 'AutoDetectBMPFlag'])
    Outputs: set(['MemoryTransceiver'])
    Functions available: ['FrontEndCommonMachineGenerator']
    Functions used: ['FrontEndCommonHBPMaxMachineGenerator', 'FrontEndCommonVirtualMachineGenerator', 'MallocBasedChipIDAllocator', 'PartitionAndPlacePartitioner', 'FrontEndCommonHBPAllocator', 'MallocBasedChipIDAllocator']
    Inputs required per function: 
            FrontEndCommonMachineGenerator: ['['DownedLinksDetails']']

alan-stokes, 2017-08-01:
fixed in https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/pull/170

Christian-B, 2017-08-03:
fixed

closed 2017-08-03 by Christian-B