SpiNNakerManchester/SpiNNFrontEndCommon #150: Setting parameter ip_tag.port to value None is invalid:

Christian-B, 2017-06-26:
ONLY when using a board not with spalloc.

Found with a synfire chain run twice without extrating data but with a reset

https://github.com/SpiNNakerManchester/sPyNNaker7/blob/master/p7_integration_tests/test_multi_call_examples/test_synfire_2_run_reset_new_pop_if_curr_exp.py

and 

https://github.com/SpiNNakerManchester/sPyNNaker8/blob/master/p8_integration_tests/test_multi_call_examples/test_synfire_2_run_reset_new_pop_if_curr_exp.py

Traceback (most recent call last):
  File "/brenninc/spinnaker/sPyNNaker7/p7_integration_tests/test_multi_call_examples/test_synfire_2_run_reset_new_pop_if_curr_exp.py", line 24, in test_run
    reset=reset, new_pop=new_pop)
  File "/brenninc/spinnaker/sPyNNaker7/p7_integration_tests/scripts/synfire_run.py", line 277, in do_run
    p.run(run_times[-1])
  File "/brenninc/spinnaker/sPyNNaker7/spynnaker7/pyNN/__init__.py", line 157, in run
    globals_variables.get_simulator().run(run_time)
  File "/brenninc/spinnaker/sPyNNaker/spynnaker/pyNN/abstract_spinnaker_common.py", line 326, in run
    AbstractSpinnakerBase.run(self, run_time)
  File "/brenninc/spinnaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/abstract_spinnaker_base.py", line 719, in run
    self._do_load()
  File "/brenninc/spinnaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/abstract_spinnaker_base.py", line 1380, in _do_load
    inputs, algorithms, outputs, optional_algorithms)
  File "/brenninc/spinnaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/abstract_spinnaker_base.py", line 885, in _run_algorithms
    executor.execute_mapping()
  File "/brenninc/spinnaker/PACMAN/pacman/executor/pacman_algorithm_executor.py", line 422, in execute_mapping
    self._execute_mapping()
  File "/brenninc/spinnaker/PACMAN/pacman/executor/pacman_algorithm_executor.py", line 438, in _execute_mapping
    results = algorithm.call(self._internal_type_mapping)
  File "/brenninc/spinnaker/PACMAN/pacman/executor/algorithm_classes/abstract_python_algorithm.py", line 45, in call
    results = self.call_python(method_inputs)
  File "/brenninc/spinnaker/PACMAN/pacman/executor/algorithm_classes/python_class_algorithm.py", line 57, in call_python
    return method(**inputs)
  File "/brenninc/spinnaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/interface_functions/front_end_common_tags_loader.py", line 35, in __call__
    self.load_iptags(tags.ip_tags, transceiver, progress)
  File "/brenninc/spinnaker/SpiNNFrontEndCommon/spinn_front_end_common/interface/interface_functions/front_end_common_tags_loader.py", line 57, in load_iptags
    transceiver.set_ip_tag(ip_tag)
  File "/brenninc/spinnaker/SpiNNMan/spinnman/transceiver.py", line 2262, in set_ip_tag
    "The tag port must have been set")
SpinnmanInvalidParameterException: Setting parameter ip_tag.port to value None is invalid: The tag port must have been set

Christian-B, 2017-07-27:
Unable to replicate on chip-power-monitor branch so suggest closing when that branch is merged in

alan-stokes, 2017-07-27:
begs the question, whats different in chip-power-monitor that fixes this issue? it doesnt touch the dynamic ports stuff to my knowledge.

alan-stokes, 2017-08-01:
well assuming its fixed now, given chip-power-monitor is in master. will be closing issue. if found again, reopen. rather suspicious why chip-power-monitor fixed it.

closed 2017-08-01 by alan-stokes