SpiNNakerManchester/SpiNNFrontEndCommon #180: HBP Server needs testing

Christian-B, 2017-08-03:
I HIGHLY recommend the HBP server is tested before release.

We accidentally found a bug that would have made it completely unusable when updated to latest code.
See: https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/pull/170

Are there others?

alan-stokes, 2017-08-03:
how to test the HBP portal on the other hand is a open ended box. more related to supporting more than 1 version of the tools for the portal. Has the smell of pandora's box

rowleya, 2017-08-15:
We can test by actually installing it on the portal system, though this requires a little bit of a finesse...

andrewgait, 2017-08-23:
I'd be happy to look at this in my "user support" sort of role if no-one else has looked at it yet... ?

alan-stokes, 2017-08-30:
by installing it on the HBP portal, don't we expose the end users to whatever cock ups we've done? what we were thinking was a test state which can test the flow without interfering with end users till we are confident it works. which is why its got a wiff of a Pandora box.

rowleya, 2017-09-11:
This has been tested using the staging server, so closing now.

closed 2017-09-11 by rowleya