SpiNNakerManchester/SpiNNFrontEndCommon #190: slow-ish database creation

chanokin, 2017-08-17:
For large networks (~200k neurons with one-to-one and convolution-like connectivity) generating the database becomes slower than loading the network itself. 

Could it be possible to switch to a higher- performance  DB manager for this cases?

rowleya, 2017-08-18:
We currently store the key of every neuron, which is probably what is really slow.  In reality, we could get away with storing the keys, and then have the visualiser (or even the reading library) work out the exact mapping for a full population.

alan-stokes, 2017-08-30:
this is true in theory. The reason for that table was that there were devices which didn't want to do that exact logic, and wanted a simple lookup instead. I think (though please correct me if i'm wrong) that there's a config param for turning off the database generation of that specific table. 

[Database]
create_routing_info_to_neuron_id_mapping = True

But you'd then need to upgrade the vis that your using to do the logic as rowley stated. 

switching to a "higher-performance DB" wont help you here, given populating DB's are not traditionally optimised, its the querying of the DB that's most focused on.

alan-stokes, 2017-12-20:
i persume this issue is now closable?

chanokin, 2017-12-20:
Nothing has really changed, right? We probably need a way of reducing what's going into the DB before closing this.

rowleya, 2017-12-21:
This is still an issue.  I am unsure, but I suspect the issue here could be resolved by:
1. Only write the keys for populations that are live output or live input.
1. Only write the base key and number of keys rather than all the keys.
1. Change the database readers to generate the full set of keys using the database information.

alan-stokes, 2017-12-21:
or using this flag in cfg?
[Database]
create_routing_info_to_neuron_id_mapping = False

given anything that requires that map, the db has the data that would allow the software to recreate it anyhows.

rowleya, 2017-12-21:
If you need the mapping, and it takes ages to create the database, people won't want to use it.  Thus having a faster solution is important, and not too hard to achieve.

alan-stokes, 2017-12-21:
yea, solution, don't build the mapping in the database, let the system which needs it build it. any of the other options is a sticky plaster. 
1. Only write the keys for populations that are live output or live input. right and when we have all pops being live outputted (or enough), they'll complain again.
2. Only write the base key and number of keys rather than all the keys. then your basically asking the user to fill the gaps, well given that the base key and n keys are in other tables, they can do that search also, so why 1/2 arse it.
3. Change the database readers to generate the full set of keys using the database information. well that basically what i'm saying, and is done via that flag, and thats what im saying is the only real scalable solution. Itll make the reader slower, but then what ya expect when your trying to do such things.

rowleya, 2017-12-21:
The three things were meant to be done together... if we can make the reader do it without writing anything (i.e. option 3 without 1 or 2), all the better!  It will make the reader slower, but it will make the writer faster, so those who want to use it get it, and those who don't want it don't get it.

dkfellows, 2021-03-26:
Some effort has been put into making things faster since then, with better transaction management and use of bulk inserts where possible. Is this still an issue?

closed 2023-09-26 by dkfellows