SpiNNakerManchester/SpiNNFrontEndCommon #200: Remove data files on upload option

rowleya, 2017-09-18:
It would be useful to remove data files when they have been uploaded to save users having lots of useless files on their system.  An additional option to keep the files should also be added.

pabogdan, 2017-09-19:
My hack revolves around adding 
```python
        if os.path.exists(reloaded_dsg_data_files_file_path):
            shutil.rmtree(reloaded_dsg_data_files_file_path)
        if os.path.exists(reloaded_dsg_report_files_file_path):
            shutil.rmtree(reloaded_dsg_report_files_file_path)
```

at the end of the __call__method, but it would be much nicer if writing to disk would be completely avoided.

(FYI shutil.rmtree deletes directory and all containing files / folders)

rowleya, 2017-09-19:
Alan correctly pointed out to me that this requires some careful thought as we reuse some data files when the user calls ```p.reset()```.  There are definitely some files that we can get rid of, and we can get rid of them all on p.end.

pabogdan, 2017-09-19:
@rowleya Then it's a good thing my synaptic rewiring does not currently have an interface for PyNN0.8, so my stuff will work as long as I stay to PyNN0.7. Good to know though

rowleya, 2017-09-19:
Except that PyNN 0.7 supports reset as well...  this is all just sPyNNaker, not specifically sPyNNaker 7 or 8.

dkfellows, 2023-09-26:
I'm guessing if you're going to do this, it ought to be a config option _defaulting to off_.