SpiNNakerManchester/SpiNNFrontEndCommon #203: C++ compiler *fix*

chanokin, 2017-09-25:
I've added the _-ffreestanding_ option to the compiler flags and the _-nostartfiles_ to the linker. This "fixes" compilation of the connection builder module, though I need to test if the binary is doing what it's supposed to.

rowleya, 2017-09-26:
Excellent - good to know!  I plan on looking at the integration of this in the coming weeks / months (probably after the summit now)

chanokin, 2017-10-02:
I checked on a brand new install in a virtual environment and the latest compiler I could find (i.e. 5.4 2016q3) and the binary seem to be doing what it's supposed to.

andrewgait, 2017-10-03:
If you want to, you can test this out on the latest version (version 6, which came out earlier this year) of the compiler - you can get it at the following link (for Ubuntu):

https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa

If you're using Windows then I believe you can get version 6 there too; as far as I can see, though, you can't get version 6 for MAC OSX yet...

chanokin, 2017-10-03:
Thanks, @andrewgait, I've tested the compilation and it still works.

BTW the new place for downloads is:
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

andrewgait, 2019-07-23:
As far as I can see both these flags are included in the relevant place in spinnaker_tools.mk so I'm going to close this issue.

closed 2019-07-23 by andrewgait