SpiNNakerManchester/SpiNNFrontEndCommon #221: armcc support in Makefile.SpiNNFrontEndCommon broken

lplana, 2017-11-13:
Compiling the SpiNNaker_PDP2 C code with armcc produces the following error:

armcc -c --c99 --cpu=5te --apcs interwork --min_array_alignment=4 -I /home/plana/scratch/gfe_tests/pdp2/spinnaker_tools/include -Ofast -Wall -Wextra -DPRODUCTION_CODE -Otime -DAPPLICATION_NAME_HASH=0xa43c1b0a -g -o build/input.o input.c
Fatal error: C3900U: Unrecognized option '-all'.
Warning: C3910W: Old syntax, please use '-E'.
Fatal error: C3900U: Unrecognized option '-xtra'.

The error seems to be related to the use of options '-Wall' and '-Wextra'. Additionally, using options '-Ofast' and '-Otime' does not seem right.

The SpiNNaker_PDP2 Makefile does *not* set any compilation flags or options. It #includes 'Makefile.SpiNNFrontEndCommon'.

Compilation completes correctly for arm-none-eabi-gcc.

andrewgait, 2019-07-23:
I think the new makefile system means that you should be using either local.mk or spinnaker_tools.mk rather than Makefile.SpiNNFrontEndCommon in this instance (it looks to me as though there is a distinction in spinnaker_tools.mk between armcc and arm-none-eabi-gcc at least for -Wall and -Wextra).  If you could try it again with either of these to see if it fixes the issue that would be good.

andrewgait, 2020-04-20:
Hi @lplana  - since you and @shrinkingviolet are using SpiNNaker_PDP2 again at the moment, does that mean this issue has been fixed, or is it still a problem?

lplana, 2020-04-20:
Hi @andrewgait. SpiNNaker_PDP2 is 'stuck' a bit in the past, so our versions of spinnaker_tools and SpiNNFrontEndCommon do *not* offer the alternative makefiles that you mentioned above. The problem might have been solved, but I cannot comment from our perspective. We're using gcc, as this works fine.

We may try to upgrade SpiNNaker_PDP2 to the latest version of SpiNNTools. If so, I'll report on this issue.

lplana, 2020-05-02:
Hi @andrewgait. I've upgraded SpiNNaker_PDP2 to the latest version of SpiNNTools and can now confirm that this issue is __has__ been sorted. With respect to that project, this issue can be closed.

lplana, 2020-05-02:
I was a bit too hasty reporting success earlier. The problem is still there. 'local.mk' doesn't look at the GNU variable at all. We should just drop __arm__ support altogether. There are no users, anyway.

rowleya, 2020-05-04:
I think the current ARM support only extends as far as SCAMP, where it can be essential for getting things the right size...