SpiNNakerManchester/SpiNNFrontEndCommon #227: CodeSubsets do not have attribute Value

Christian-B, 2017-11-15:
Error reporting such as:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/interface/interface_functions/chip_provenance_updater.py
__call__

Fill in errors with 
error_cores.values()
Where error_cores comes from the 
https://github.com/SpiNNakerManchester/SpiNNMan/blob/master/spinnman/transceiver.py get_cores_in_state(..

Which returns a:
https://github.com/SpiNNakerManchester/SpiNNMachine/blob/master/spinn_machine/core_subsets.py

Which does not have values() method.

alan-stokes, 2017-12-20:
this still a issue?

rowleya, 2017-12-21:
It would appear so - I guess it works because the cores haven't failed... should be fixed at some point though

Christian-B, 2019-02-14:
Fixed with https://github.com/SpiNNakerManchester/SpiNNMachine/commit/f7565a339e760ecbef75c4aecc9e98977469eb55

closed 2019-02-14 by Christian-B