SpiNNakerManchester/SpiNNFrontEndCommon #250: Increasing size of recorded V causes DataSpeedUpPacketGatherMachineVertex to fail

Christian-B, 2017-12-08:
Script:
import spynnaker8 as sim
from spynnaker8.utilities import neo_convertor

# number of neurons in each population
n_neurons = 8

simtime = 1000

sim.setup(timestep=1.0, min_delay=1.0, max_delay=144.0)

spikeArray = {'spike_times': [[0]]}
stimulus = sim.Population(1, sim.SpikeSourceArray, spikeArray,
                          label='stimulus')
pop =  sim.Population(n_neurons, sim.IF_curr_exp, {}, label='chain')
pop.record(["spikes","v"])

sim.Projection(stimulus, pop, sim.OneToOneConnector(),
               synapse_type=sim.StaticSynapse(weight=5, delay=2))
loop_connections = list()
for i in range(0, n_neurons):
    single_connection =  (i, ((i + 1) % n_neurons), 5, 2)
    loop_connections.append(single_connection)

sim.Projection(pop, pop, sim.FromListConnector(loop_connections))

sim.run(simtime)
neo = pop.get_data(["spikes","v"])
spikes = neo_convertor.convert_spikes(neo)

sim.end()

print spikes

-
Master as of dec 7th 8pm
spin 3 board
windows

-

Above script works.

BUT
Changing number of neurons to 9 and it fails.
Changing the simtime to 1200 and it also fails
Adding recording of both gsyn has no effect,

DataSpeedUpPacketGatherMachineVertex block as line 330
it is the First attempt to get data that fails.
It is placement 0,0,2

rowleya, 2017-12-08:
Did you happen to capture the error message?

Christian-B, 2017-12-09:
It is the one thrown by the DataSpeedUpPackageGathererMachineVertex about line 337.
I added extra code count the succesful receives in that block and it was always zero

Christian-B, 2017-12-11:
correction: not spin 3 board
RATHER spinnaker.cs.man.ac.uk but REMOTELY from London

alan-stokes, 2017-12-20:
not any more. reopen if proved otherwise

closed 2017-12-20 by alan-stokes