SpiNNakerManchester/SpiNNFrontEndCommon #261: Simulation running forever cannot be restarted after reset

georghinkel, 2017-12-21:
The reset method in the _AbstractSpinnakerBase_ class stops the cores, but does not change the state. Therefore, the simulation cannot be continued afterwards. Furthermore, when I add a line to reset the __state_ to _INIT_, then the next run fails with a SpinnmanIOException that a pause/stop command is expected. Instead, a database confirmation message is sent.

I used the following code to run the simulator in _run_forever_ mode, suspend and run again:

```
import pyNN.spiNNaker as Frontend

# ... setup, create network

from spinn_front_end_common.utilities import global_variables as vars
from spynnaker.pyNN.abstract_spinnaker_common import AbstractSpiNNakerCommon
simulator = vars.get_simulator()

# Run forever: There is no way to run the simulator in run_forever mode through PyNN, so I use the AbstractSpiNNakerCommon API
AbstractSpiNNakerCommon.run(simulator, None)

# Suspend: I found that reset does most of what I want here: It stops the application but does not change the network or mapping
Frontend.reset()

# Resume: This throws an error message that the simulation is still running. 
# When I patch that in the reset command, another error message is thrown that an EIEIO command pause or stop was expected but not received
AbstractSpiNNakerCommon.run(simulator, None)
```

I am using the PyPy version, currently.

Is this use case officially supported (I guess not), is my approach to suspend the simulation using _reset_ a good one? I have also tried to issue another run command with zero time just in order to get the pausing behavior but that also did not work. What I want to achieve is a simple suspend, that means no changes to the network and no changes to the routing table.

rowleya, 2017-12-21:
This is an interesting use case, though not one we had thought of.  Reset has a specific meaning here, so I can see why it doesn't work.  What we could do with instead would be a pause; however the difficulty is ensuring that all cores are then in the same state at the pause point i.e. that they are all at the same time step (assuming this is a timed simulation).  This will take some thought...

alan-stokes, 2017-12-21:
just to add my 6 bits: 
you say: 
# Run forever: There is no way to run the simulator in run_forever mode through PyNN, so I use the AbstractSpiNNakerCommon API AbstractSpiNNakerCommon.run(simulator, None) 

given you've gone below the official API of the front end your using, you've already got yourself at a point where we could happily say you broke the agreement, what did you expect?! But in the act of trying to solve your issue. 

there is a pynn.run() which will run forever, by handing control back to the top script once the system has started. Its then on your head to wait till your ready to end (possibly using a busy wait and a command prompt), and then call pynn.end().

Its not true PyNN, given PyNN is batch mode mentality, but it is still within the interface we officially support for PyNN. Using AbstractSpiNNakerCommon functions is only suppose to be used by FrontEnd developers, not front end users. 

If you want to rerun something at this point. Your "official" path is then to do a pynn.setup() and rebuild your graph and then run your script again. This is using the API in the way it was intended. pynn.reset and the rest of the interface functions are designed to work with pynn.run where there is a given runtime, so using them in a run forever mode has very dubious meaning.

This all said and done, you can emulate a run forever using a while(true): p.run(x), which keeps you in the offical pynn interface as well. 

As a lesson, when you start calling none official functions, you really should ask yourself beforehand, is this really what i should be doing?

georghinkel, 2017-12-21:
Hi Andrew, hi Alan,

first of all thanks for your comments!

I am well aware that I can't really expect that everything works immediately when I am not using official APIs, I am rather interested to see what functionality we require to integrate spinnaker into the HBP Neurorobotics Platform is already there in order to estimate an asynchronous integration.

Currently, we do have a first integration of spinnaker where we do more or less exactly do this while (True): p.run(a small x) loop, but this had caused quite an overhead when we did that first about almost two years ago. For this reason, we wanted a "free run integration" with a goal to reduce communication overhead, but I was not exactly sure what we mean by that: We originally planned to leave the spinnaker board alone as much as we can except that we communicate spikes with it. That's why I was trying to set the simulation in this _run_forever_ mode. To be honest, after we recently revived the spinnaker integration, the performance actually seems quite good to me (but also for that, we use a non-standard API, namely _spinnaker_get_data_), so I am actually not sure that a true free run integration is actually a clever idea. From your answers, it sounds to me that this run_forever mode is not desirable, but the synchronized control of repeatedly running _pyNN.run_ is.

Best,

Georg

alan-stokes, 2017-12-21:
hi Georg

you say:

while (True): p.run(a small x) loop, but this had caused quite an overhead when we did that first about almost two years ago.

it would be interesting to see what this overhead is that is believed to be too much. I'm betting its the data extraction from the auto pause and resume, but the most up to date master has functionality now that gives you a 15 times speed up in extraction of data (and that's only touching the surface of how good it can go, we're still exploring its capabilities) to that might put you back into "acceptable overhead" whatever that means.

We've added a lot more functionality into being able to change poissions rates during running scripts, and there's always the live input and output functionality you can depend upon for spikes.

The spinnaker_get_data() call is just a call to avoid using the Neo objects, which we recognise take up loads more processing time to get the data into its format, and for people to use it for getting data out.

This all said and done. Running forever very much removes a lot of safety code we've added to make your simulations run smoothly, provenance gathering, auto pause and resume for data extraction, flow control. Running via the pynn interface is our expected interface, the rest are rather "do at your own risk". 

This all said, i don't think running forever and reset is going to helps you out in relation to overhead. Reset by PyNN's definition is go back to that state it was in at t = 0, so how that works in spinnaker is we throw away the executable and on the next run we reload all the binaries, and app data etc again, so your overhead cost is going to be much more huge than by staying in the API calls (pulling v,gsyn,spikes is maybe a few meg at most, the app data can be in giga bytes at least). We don't do remapping (so placement etc) unless you've changed your network. But the data itself has to be reloaded, as any plasticity has to be reset, the neuron parameters need resetting etc. 

this all said and done, there is also a run_until_complete() call in the GFE which allows you to be partly infinite run, but it expects the c code to eventually get to a pause/end state and locks up the top script until that occurs. I don't think its exposed to the PyNN front end, but it might not be too hard to do so, but then you need a way of stopping the c code. Now there is a scp command the cores will recognise as a kill command, but p.end() / reset() will do almost the same job, and its not 100% going to work, as its assuming the cores are idle enough to process the interrupt.

Having done some digging, even our pynn interface isnt consistent between 0.7 and 0.8/0.9. In 0.7 run(None) does indeed set up for infinite run, our 0.8/0.9 version cant do the same, as it tries to tie into PyNN source more and that cant handle None. Nor is there any run-forever() in the external devices module of 0.8/0.9. oops. We could easily re-expose that though......... but as i've said, i think your barking up the wrong tree here if your looking for performance increases.

Why do you want to reset?

georghinkel, 2017-12-21:
I just used reset because I found that it did not that much more than stopping the application (though perhaps I underestimated the overhead that pyNN adds to this call, I first wanted to see whether it works or estimate how much needs to be done to get it working). 

But I guess you pretty much convinced me that staying with the pynn interface is the better option and performance-wise, it is much better than what we had two years ago. If you say you have even better results with the current master, then I will wait for that to be released as PyPy package :)

Thanks again for your replies, I will close this issue now because the actual problem is solved

closed 2017-12-21 by georghinkel