SpiNNakerManchester/SpiNNFrontEndCommon #266: Extra monitor support assumes existence of Ethernet chips when using an allocated machine

rowleya, 2018-01-09:
There are a few Ethernet chips in the machine that have flaky default monitor cores.  Unfortunately, if you get one of these machines from spalloc, it won't work, as it assumes that the ethernet_connected_chip exists, whereas actually this is just the local ethernet chip that *should* exist, but might not.  This is really a fault in the machine, which is being investigated, but it would be helpful if the tools could work around this issue too.

alan-stokes, 2018-01-12:
supporting such functionality is a rather big pigs dinner. Its not just extra monitor stuff, but mulitboard live io in general. 

for speed up, youd need to modify the fixed router to send things down or up to the nearest working Ethernet, and youd need to change all the lgoic to test for these dead boards and try other ones for data extraction. 

for live io, its even more piggy, as sdp packets etc go to the nearest ethernet x and y. which would need to be adjusted and the same level of logic as the speed up stuff would need to occur for choosing tags etc. 

my view point. its faster, less effort,and more clean from the tools viewpoints if those boards are removed from spalloc. Given the transient nature of this bug (and that luis is working on fixing it).

rowleya, 2018-01-13:
There is already a SpiNNFrontEndCommon branch (fix_missing_ethernet_chips) that copes with this.  A quick analysis (searching for nearest_ethernet in the code) shows that the only place that isn't handled is sPyNNaker, and that looks easy enough.

The current solution is, when looking at nearest_ethernet_x, nearest_ethernet_y, check if the chip exists.  If it doesn't use 0, 0.  When doing resource allocation, if the user specifies a board_address that doesn't exist (i.e. that chip is missing) it will fail as expected.  If the user just specifies a tag, it will try the nearest Ethernet chip first.  This is already set up to either fail because there are no tags left on that chip, or else fail because the chip is not there.  In that case, it performs normal resource allocation, which is to say that it just searches for another Ethernet chip with tags available and uses that.  This will work in general because all chips can talk to all Ethernet chips; it isn't as efficient, but it isn't wrong.

The only other outstanding issue is the current FixedRouteRouter.  This algorithm is flawed in general from what I can see, because it doesn't even check the edges of the graph, but instead assumes that there will be a destination on the Ethernet chip.  A possible solution is to use multicast instead of fixed routes in general, as then the routing is fairly simple.

alan-stokes, 2018-01-15:
edges of the graph? its not meant to be using the graph much at all. its meant to create a comms fabric where theres a thing on each ethernet chip. I dont think using the mc always is a best option, its a kludge for when there's a unclean machine setup. 

and you have not satisified the software fully, spike io? arguely the system should adapt and adjust its comms to the working Ethernet chip, but it wont from what i can see so far. SDP packets from cores wont go to the correct places, as they work off nearest x and y on machine. Say IO_STD, SDP live buffers, etc etc.  

these "fixes" are not going to achieve a stable stack when dealing with missing Ethernet chips, and the work to make the ENTIRE stack work when missing Ethernet chips is more work than its worth when removing the broken board will rectify the problem.

rowleya, 2018-01-15:
Fixed Route is a special entry in the router that works like Multicast except that the key doesn't need to be specified as there is only one FR entry in the table.  Thus you can use it for more than just communication with the Ethernet chip.  The code that inserts edges to extra monitor functionality adds a Fixed Route edge between the extra monitor core and a data gatherer vertex.  The code that routes Fixed Route edges ignores them and assumes that a) there is a source on every chip and b) that the destination of every source is the Ethernet chip of the board, whether there is one or not!  Thus it is not really a fixed route router.

SpikeIO should work correctly in master with missing Ethernet chips I believe.  This is because the bit of code that adds edges to the LPG vertices from those that send things tries the nearest Ethernet chip and if it can't find it then just uses 0, 0 (as this will always exist according to Luis).  SpikeInjector will work so long as there are some tags available, and there can only be tags on Ethernet chips.  Any Ethernet chip is fine for the tags in this case.

I believe that these changes *are* enough to make the entire stack work when missing Ethernet chips, as these are the only bits of code that try to look for them.

alan-stokes, 2018-01-16:
"Fixed Route is a special entry in the router that works like Multicast except that the key doesn't need to be specified as there is only one FR entry in the table. Thus you can use it for more than just communication with the Ethernet chip. The code that inserts edges to extra monitor functionality adds a Fixed Route edge between the extra monitor core and a data gatherer vertex. The code that routes Fixed Route edges ignores them and assumes that a) there is a source on every chip and b) that the destination of every source is the Ethernet chip of the board, whether there is one or not! Thus it is not really a fixed route router."

i concede that point......... i hadn't thought of it in those terms. 

"I believe that these changes are enough to make the entire stack work when missing Ethernet chips, as these are the only bits of code that try to look for them."

i don't believe that's true, io_std is a perfect example where this doesn't hold, as io_std uses the on machine nearest ether x and y, which to my knowledge will still be pointing at a bust chip. The same goes to any SDP packet being sent out from a core (aka live buffer, which i also concede don't work for software slippage reasons) which uses the same logic. 

if we add code that reconfigures the chips nearest x and y, i think id be happy to accept that it would work, which makes these proposed changes for just going to 0,0 go away, as then the area code is adjusted accordingly. Currently your dumping on the 0,0 because that's always there, but youre opening to failure due to overloading that itself. Readjusting the area codes so that its spread out would i beleive be more suitable, then your check is just for its nearest ethernet x and y and no buffer logic is required. 

I do note that yes, fr router is flawed in this respect.

rowleya, 2018-01-16:
I agree that IO_STD is broken in this model, and understand that you would like the machine to update the nearest Ethernet chip when booting.  This could be made an issue in spinnaker_tools therefore...

alan-stokes, 2018-01-16:
doesn't need to be when booting. but that would also solve the issue, at which point our software need not reconfigure itself. 

I was proposing our software during machine detection did so.

alan-stokes, 2018-01-16:
by having our stuff do it, you end up with reduced number of ethernet connected chips and bigger area codes. it also allows us to decide how this mapping should occur. whereas doing in boot is forced to change spinnaker tools every time, whereas as a sepreate algorithm, its easily replaceable

rowleya, 2018-01-16:
OK, that is also a reasonable solution.

Christian-B, 2021-03-22:
Did we not remove any boards where the local ethernet chip does not exist?

rowleya, 2021-03-22:
We have indeed; such boards would make things on spalloc quite hard!

closed 2021-03-22 by rowleya