SpiNNakerManchester/SpiNNFrontEndCommon #282: Incorrect use of wait_for_cores_to_be_in_state

rowleya, 2018-02-20:
The SpiNNMan function wait_for_cores_to_be_in_state (https://github.com/SpiNNakerManchester/SpiNNMan/blob/master/spinnman/transceiver.py#L2020) uses an app id, so it assumes that all cores in the same app id are specified.  This is quite often used incorrectly in SpiNNFrontEndCommon, where a subset of the cores of the application are passed in, and thus the count function used in the function is wrong.

Either the calls should use different app ids for different core types, or the SpiNNMan function should be changed to make it work in these situations (hard).

alan-stokes, 2018-02-20:
your basically proposing a executable type id?

alan-stokes, 2018-08-29:
so im just going to throw this question at ya. 
How many app_ids do we have in a sim? coz my last check was 1, as everything lives in the same graph these days. 
Now if im right, then the only way this becomes a problem is if your running more than 1 sim on the same board, which we said was a error during normal operations. 

I note that the demo thingy bypassed this so you could run suduko and ..... whatever the other one was, on the same board, but that was a special case. 

if i'm wrong and there's many app ids, then i bet there's a lot of niggly issues in the spinnman and spinnaker tools counts, at which point id say the fix is to modify the count monitor code to handle a app id as well. Which would make the code in SpinnMan to just push the appid it already has at this point down to the count process.

alan-stokes, 2018-08-29:
ignore above. relised it already does

alan-stokes, 2018-08-29:
yea, having retraced all this and got to the same point as you did when we first checked this. Yep a executable type app id sounds the sensible approach. its chip_runtime_updator thats the killer. as that's the only one that goes for a specific exeuctable type core count.

it might be worth checking if its this function thats fundermentally going boom

dkfellows, 2023-09-26:
Is this still relevant?

Christian-B, 2023-09-26:
YES! The code is buggy.
It compares
self.get_core_state_count(app_id, cpu_state)
with
len(all_core_subsets) 

that these to match is by chance

Christian-B, 2026-04-07:
application_runner usage is potential dangerous as it uses the global FecDataView.get_app_id() and then calls the method in a loop based on FecDataView.get_executable_types().items()

It happens to work because FecDataView.get_executable_types() only has one value

Christian-B, 2026-04-07:
run_system_application is called three times
- compression
- neuron_expander
- synapse_expander

All three use View.get_new_id() so already have a unique app_id