SpiNNakerManchester/SpiNNFrontEndCommon #283: Database is written in every run

georghinkel, 2018-02-20:
If you repeatedly call run, e.g. by repeatedly calling run through the PyNN API, the DatabaseInterface algorithm is always executed which in turn always writes the database. 

For applications that call run repeatedly (such as the Neurorobotics Platform developed in HBP SP10), this causes a lot of overhead for times between run calls. For a small test network, it took about 3s where we are calling run with about 100ms as time argument. It would be great to write the database only whenever this is really necessary in order to reduce the overhead and therefore improve the ratio between actual simulation time and simulation time plus overhead between repeated run calls.

alan-stokes, 2018-02-21:
hopefully fixed in 
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/pull/284

andrewgait, 2020-04-20:
Looks to me like this has been fixed; please open again if it still needs work.

closed 2020-04-20 by andrewgait