SpiNNakerManchester/SpiNNFrontEndCommon #293: Change recording calculations

rowleya, 2018-03-02:
Recording calculations may be better if we consider the size of a "recorded element" in the buffer space of each recorded region.  This could work with the vertex having the following interface:
```
def get_size_of_recording_elements(self):
    """ Get a list of the sizes of an element of recording, \
       where an element is an atomic unit of recording
    """

def get_n_time_steps_for_n_recorded_elements(self, n_elements):
    """ Get the maximum number of time steps that can be \
       executed to produce the given number of recorded elements.
    
    :param n_elements: The number of recording elements that will be made
    :return: The number of time steps that will produce n_elements recordings
    """
```

The system would then work as follows:
1. Add the space to record one element (or a user-configured number of elements) of each recording region to the space required by each vertex.  Each vertex should then *not* request any SDRAM for recording explicitly (this is better than the current system, where we have a minimum buffer space).
1. Work out the amount of SDRAM available for recording on each chip after every element has run as is currently done.
1. On each chip, work out the total space for recording one element of each recorded region for each vertex on that chip.
1. Divide the space available by the total space required for one unit of recording.  This is the number of recording units that can be stored, n_elements.
1. Ask each vertex how many time steps n_elements will run.
1. Use the minimum such value to determine the number of steps in each auto pause and resume cycle.

I believe that this will resolve some of the issues in the current calculation, but if there are any other ideas, please discuss in this ticket!

alan-stokes, 2018-03-07:
sounds doable. would need to ensure min is a whole number of cycles, not some fraction.

Christian-B, 2018-03-07:
I highly disagree with this approach.

In my opinion the cleanest approach is to ask each vextex.

1. Amount of extra SDRAM for each additional timestep

2. Amount of fixed SDRAM required regardless of number of time_steps.
if the vertex does not use exactly the same amopunt of SDRAm every time_step this value should included a reserve for the worst error in 1

Then for each chip 
1. Allocate the fixed sdram and compute available SDRAM
2. Total all the extra SDRAM per time_step required
3. Divide available SDRAM by the total per timestep to generate the max time steps.
4. Allocate available SDRAM for the max timestep even if not used in run 1

For whole graph run for lowest max timestep per chip

alan-stokes, 2018-03-07:
the "allocate the fixed SDRAM and compute available SDRAM" is one of the issues here. if you don't allocate anything to the partitioner for this. its in its right to partition in such a way as to leave nothing for recording. so allocating 1 "atomic element" which is the end user definer minimal auto pause and resume step gets you around this issue. i think. 

and regardless of what logic we do, we still disregarding the issue that these free SDRAM might not be one continious chunk. so theres some work there that we need to consider

Christian-B, 2019-07-08:
Is this still relevant after variable sdram?

rowleya, 2019-07-08:
No, I suspect this was meant to be something similar, but this is now obsolete

closed 2019-07-08 by rowleya