SpiNNakerManchester/SpiNNFrontEndCommon #298: reset doesn't reload data correctly

rowleya, 2018-03-18:
The following sequence doesn't work correctly: run, change parameter, run, reset, run.  The region reloader is called following a reset, but the vertices don't recognise that they need to be reloaded, as they have already been reloaded in the second run.

alan-stokes, 2018-03-19:
as found from pynn api
reset(annotations={})
Reset the time to zero, neuron membrane potentials and synaptic weights to their initial values, and delete any recorded data. The network structure is not changed, nor is the specification of which neurons to record from.

i think, which is up for debate, that this implies that the changed param is wiped away and whatever it was set to before the first run should be reloaded. 

If that is the case, its still works, as we load the original dsg files, not the reloaded ones. I do admit that currently this emans the python neuron params are not matched, and that's a different problem. But it would mean run, change, run, reset, run is currently correct from the c code perspective.

alan-stokes, 2018-03-19:
maybe @apdavison can clear up what the expected behaviour is?

apdavison, 2018-03-19:
Ah yes, another case of "forgetting to update the docstring when you change the behaviour..." :-(

So:
  - parameter values should not be changed on reset
  - data should not be deleted on reset, rather we start recording to a new neo.Segment

https://github.com/NeuralEnsemble/PyNN/commit/78a43d7a026d6109f8b9ac1844a9e5062ce89d5b

alan-stokes, 2018-03-19:
Hi @apdavison . 
myself and christian are still discussing. 
so are you saying that reset sets the time to 0. BUT leaves everything that's occurred during whatever runs unchanged aka weight changes, structural plastic changes, new pops added between runs (if we supported them), param changes between runs, poisson rate changes etc. OR do they all get set back to the parameters states that they were in at the time = 0 (aka before the first call to run())? (and what do ya do with added pops between runs......(if anyone supported that). 

so me and christian have broken it down into 4 possible states of change. 
1. back to what was in the constructor (aka before any sets were done to the pop/proj).
2. as 1, but includes any sets that were done before the first call to run, but ignores sets after the first run.
3. as 2, but includes all sets whatever run it was called at. 
4. left as whatever state the machine is in, after the last run call. 

myself thinks it should go to 2. and christian personally thinks that it should go to 3.

apdavison, 2018-03-19:
3.

The idea is that network _structure_ (which includes neuron parameters, synapse parameters, connectivity) should be unchanged.

The only things that are reset are state variables (anything which changes _during_ a `run()`), which includes synaptic weights if they are subject to a synaptic plasticity rule.

Structural plasticity is a bit tricky. This is not formally supported by PyNN yet, even if sPyNNaker can support it, so I haven't considered how `reset()` should apply in that case. Any thoughts?

alan-stokes, 2018-03-19:
ok, so that makes our lfie much easier, given having dufg into the code, we are doing 2.5. aka, we reload the intial at t=0 of everything, but when you do a set after a reset, we reload all other sets as well. which is well bust. 

that said, changing to 3 means just doing a full dsg generation after reset, which is a few lines in abstractSpiNNakerBase. 

on structural plastic: ill ask @pabogdan

alan-stokes, 2018-03-19:
fixed in https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/pull/299

pabogdan, 2018-03-19:
I choose to rely on the quote that 
    
> the only things that are reset are state variables (anything which changes during a run()) 

Using this definition, the correct behaviour would  be to restore the initial connectivity - the connectivity is part of the state. Traditionally, the connectivity would be an element of the state which didn't change, now we're allowing it to evolve.

Looking further into the future, if I manage to get neurogenesis up and running then a ``reset()`` here would mean deleting newborn neurons and "re-creating" the ones that died.

rowleya, 2018-03-19:
See comments in #299 - a full DSG regenerate is also not quite right.

rowleya, 2019-07-05:
Now fixed in #450

andrewgait, 2019-07-23:
This was definitely fixed in #450, so closing this issue.

closed 2019-07-23 by andrewgait