SpiNNakerManchester/SpiNNFrontEndCommon #333: Can't restart simulation after error

rowleya, 2018-06-29:
If a simulation errors at points outside of _run in abstract_spinnaker_base.py, it is impossible to restart it.

Fix:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/interface/abstract_spinnaker_base.py#L1211

Change shutdown -> stop

gblomqvist, 2020-05-19:
Is there a reason this hasn't been fixed yet that isn't mentioned here?

alan-stokes, 2020-06-04:
im not sure what the error is here. 

this mini example shows that the premise is not valid. The below code allocates 1 board, makes the partitioner fail and then allows a recovery with only 16 cores. It runs to completion.

`front_end.setup(
    n_chips_required=1, model_binary_folder=os.path.dirname(__file__))

# Put HelloWorldVertex onto 16 cores
total_number_of_cores = 1000
for x in range(total_number_of_cores):
    front_end.add_machine_vertex_instance(
        HelloWorldVertex(n_hellos=10, label="Hello World at {}".format(x)))

try:
    front_end.run(10)
except Exception as e:
    front_end.setup(
        n_chips_required=1, model_binary_folder=os.path.dirname(__file__))
    total_number_of_cores = 16
    for x in range(total_number_of_cores):
        front_end.add_machine_vertex_instance(
            HelloWorldVertex(n_hellos=10, label="Hello World at {}".format(x)))
    front_end.run(10)`

alan-stokes, 2020-06-04:
so can we have clarification on what the actual issue is

rowleya, 2020-06-05:
I can't remember how this comes up, but from analysis, the issue is related to the simulator state.  It seems that the simulator is left in a state where it is not marked as stopped unless stop is called.  I think however that recent changes mean that calling setup without calling stop now works, which might be why you are not getting the error any more.  If that is true, this may not be necessary any more.

gblomqvist, 2020-06-05:
I'm just an end user, I don't actually know if this is still an issue. I just assumed that it was since this issue was still open and since I've encountered the bug myself in different ways. All I can say is that the latest release (5.1.0) appears to have this bug. E.g., the following code running with 63 available cores (on a SpiNN-3 board) crashes without the fix proposed above:

```
import spynnaker8 as sim

sim.setup()

sim.set_number_of_neurons_per_core(sim.IF_curr_exp, 1)
sim.Population(100, sim.IF_curr_exp)

try:
    sim.run(1)
except Exception:
    sim.setup()
```
But I'm guessing this code is like an end user variant of alan-stokes's code (both exhausting the number of cores available, if I'm not mistaken).

However, even with the fix proposed above, it appears that it still can't recover from a SpinnmanTimeoutException like the following:
```
spinnman.exceptions.SpinnmanGenericProcessException: 
     Received exception class: SpinnmanTimeoutException 
     With message: Operation SCPCommand.CMD_RTR timed out after 1.0 seconds 
     When sending to 0:0:0
     Stack trace: []
```
The operation timed out can vary. I've seen:
"SCPCommand.CMD_RTR"
"SCPCommand.CMD_AR"
"SCPCommand.CMD_READ"
"SCPCommand.CMD_SIG"

This issue might also be fixed in the current version, but I thought I'd let you know.

alan-stokes, 2020-06-08:
id be inclined to say that master has this solved. given the recent changes to allowing setup to be called before end (as @rowleya stated). Given that we're gearing up for a release in the near future, id say that ya can either wait till the release is done (i am not putting times on that though, i know my history on that) or switch to a developer version of the tools and see if the issues persist. 

developer software version can be found here and will likely take between 1h to a day to complete:
http://spinnakermanchester.github.io/development/devenv.html

I'll keep the issue open, until @gblomqvist clarifies that the bug no longer exhibits itself (in whatever process he wishes to do to verify this). 

But i will remove the blocker and bug tags.

gblomqvist, 2020-06-29:
I've run some tests using the latest development version and, as far as I can tell, this is no longer an issue. It should be noted however that I've not been able to reproduce the "SpinnmanTimeoutException" scenario I mentioned above. In any case, I suggest to treat the issue as resolved until proven otherwise.

alan-stokes, 2020-06-29:
the timeout exception could be anything. I'm currently aware of a possible timeout issue in the bog standard data extraction stuff. but it exhibits itself as a infinite loop, not a error being raised. Changes since the last release include a new IO protocol for mass data extraction and data loading, as well as a mother load of bugs that, that forced us to deal with. So ANY of those could have been your issue. 

I'll close the issue now. if you get other bugs, feel free to open new issues.

closed 2020-06-29 by alan-stokes