SpiNNakerManchester/SpiNNFrontEndCommon #334: ReverseIPTagMulticastSourceMachineVertex._install_virtual_key(n_keys) off by 1 error

arrenglover, 2018-07-12:
The check between the ``n_keys`` and ``max_key`` on line 260 of SpiNNFrontEndCommon/spinn_front_end_common/utility_models/reverse_ip_tag_multicast_source_machine_vertex.py doesn't allow for the *actual* maximum number of keys used.

For example, I want to use the key range 0x00000000 to 0x000FFFFF which is 1048576 keys. I create my reverse_ip_tag_multicast_source_machine_vertex(n_keys=1048576, ...). The function ``self._mask, max_key = self._calculate_mask(n_keys)`` on line 257 correctly returns: ``max_key: 1048575, mask: 0xfff00000``. However the following check for validity on line 260 cannot be successfully passed as 1048576 > 1048575.

I modified line 260: ``if n_keys > max_key + 1:`` and my application ran correctly without crashing, however I cannot say if there are any issues to doing this.

rowleya, 2018-07-12:
This sounds correct to me - there is an error if you have more keys than the maximum + 1 since this is 0-based.

arrenglover, 2019-07-08:
@rowleya - was this issue fixed? or ignored? I have a patch file that I have to continually apply to fix this issue and would like to know if I can stop using it.

rowleya, 2019-07-08:
I think this was closed in error...

rowleya, 2019-07-08:
Should be fixed by #453 - I just removed the check since the calculation should be correct!

arrenglover, 2019-07-08:
okay great! thanks! looks like it's time for another code update for me then.

closed 2023-09-26 by dkfellows