SpiNNakerManchester/SpiNNFrontEndCommon #336: ReverseIPTagMulticastSource should produce a warning if it misses packets

rowleya, 2018-07-19:
If the ReverseIPTagMulticastSource receives data too late for it to be sent, it should raise a warning (i.e. this should be included in provenance data which is then reported if > 0).  This warning could include a recommendation to increase the size of the buffer to be used, or to run the simulation for shorter periods between pause cycles.

andrewgait, 2019-08-01:
Is this not already done here: https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blame/master/spinn_front_end_common/utility_models/reverse_ip_tag_multicast_source_machine_vertex.py#L723-L730 ?  (Admittedly the warning isn't as comprehensive as suggested...)

rowleya, 2019-08-01:
Yes, I think this is correct; plus as it now uses auto-pause-and-resume, it probably doesn't need it any more anyway.

closed 2019-08-01 by rowleya