SpiNNakerManchester/SpiNNFrontEndCommon #343: 280?

Christian-B, 2018-08-28:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/interface/buffer_management/buffer_manager.py  line 429

What is the 280 and is it correct.

@alan-stokes says:
well i was wrong.  without scp header. its 272 bytes. (found from 
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/utility_models/data_speed_up_packet_gatherer_machine_vertex.py#L73)

@rowleya says:
I suspect this is the true maximum size of the data in a UDP message sent to SpiNNaker (=256 bytes from the SCP data field + 3 x 4 bytes from unused SCP arguments + 2 bytes from the cmd header + 2 bytes from the sequence number + 8 bytes from the rest of the SDP header). There might be an issue in using that additional 8 bytes from the SDP header however as the packets will be forwarded on within SpiNNaker as SDP packets, so 280 might be a bit too big and 272 might be the more accurate number.

rowleya, 2018-08-28:
It looks like @alan-stokes and I agree - it should be 272!  280 was because we thought we could remove the SDP headers as well, which is believable until you realise that SCAMP will wrap the UDP packet in an SDP packet which needs the headers.

alan-stokes, 2018-08-28:
i agree with rowley. given that my number comes from the speed up data out, which was deduced completely separately to the buffer manager functionality this issue is linked to. so lets all agree its 272. Add it to the issues with the buffer manager

Christian-B, 2018-08-28:
So if we change to 272 do we do this just on the python side in buffer manager or also in the C
for example:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/c_common/models/reverse_iptag_multicast_source/reverse_iptag_multicast_source.c  line 69

Christian-B, 2018-08-28:
I assume we should change buffermanager.py (python code)

rowleya, 2018-08-28:
We definitely should change this in the Python code.  In the C code you point to above, it does no harm, but equally may as well be changed also.  If this is used in any other code, it would be worth checking to see what harm it might do.

dkfellows, 2018-08-28:
I was about to ask how EIEIO implementations send and receive packets, but I don't really want to know. For (host side) receiving, what matters is "is the buffer big enough?" and for sending, it's "can I get the message through all the layers of SDP and so on to where it is expected?"

Since I think the host only actually sends small EIEIO packets (less than 280) to SpiNNaker, it isn't very important. Bulk data input is done differently.

rowleya, 2018-08-29:
This depends on the thing being buffered.  This is mostly used by the SpikeSourceArray in PyNN, and generally supported through ReverseIPTagMulticastSource.  The sending of data is done in 2 parts:
1. At the start, the buffer memory is filled (or all the messages are loaded).  This is done through transceiver.write_memory which writes the packets as raw data.  This is the only time when 280 is really used, and since the C code on the other side can accept 280 byte packets, this should work correctly.  It isn't ideal that it doesn't match up with the normal packet size limits, but it also shouldn't go wrong.
1. When the buffer has a certain amount of space on the host, it will ask the host for new data.  This will attempt to create packets filled with as many keys as possible, with a restriction that the keys must share a timestamp.  Because of this restriction, the size of the packets will be very data dependent - if a user has ~64 neurons in a SpikeSourceArray and all of them have the same send_times, this will fill packets every time.  Note that this uses a different constant for the size of the packet of 256 (which is probably a bit too small but at least it should work correctly).

rowleya, 2018-08-29:
In summary, using a single constant of 272 is probably the right answer here every time.  It isn't broken as it is, but equally it isn't consistent and so this adds to the confusion of how it works.

Christian-B, 2018-08-29:
280 found in 
reverse_iptag_multicast_source.c
And BufferManager.py

But a full search of all *.c and *.py files does not find it anywhere connected to SpikeSourceArray

alan-stokes, 2018-08-29:
thats coz the spike source array is a inheirted class of RIPMCS and doesnt do much but change spikes into the buffer format. and its the buffer manager that decides how to send data to spinnaker. so the RIPMCS shouldnt contain anything about packet sizes.

dkfellows, 2018-11-05:
So it is 272? Great. We'll use that consistently going forward.

closed 2018-11-05 by dkfellows