SpiNNakerManchester/SpiNNFrontEndCommon #346: Minor BufferManager weirdness

Christian-B, 2018-08-30:
Not bugs just weirdness.

The are two variables.
_extra_monitor_cores which is a list of These
_extra_monitor_cores_by_chip which is a map for core to these same cores.

Why not just use
_extra_monitor_cores_by_chip.values ?

Christian-B, 2018-08-30:
def _request_data(self, transceiver, placement_x, placement_y, address,
                      length):

Why pass in self.Transceiver to a private method?
and then have to do # pylint: disable=too-many-arguments

alan-stokes, 2018-08-31:
extramonitor cores. i would assume thats first pass vs optimisations. Id just check beforehand where the list is used. I bet the workflow manager includes a few places where only the list is handed over.

_request data txrx. Phft, no idea, would need to have line numbers and look at what calls it.

Christian-B, 2019-02-14:
Possible optization but best done after release.

Christian-B, 2021-03-22:
_extra_monitor_cores aka MemoryExtraMonitorVertices is used elsewhere for example HostExecuteApplicationDataSpecification so has to be generated anyway so why not use.

closed 2021-03-22 by Christian-B