SpiNNakerManchester/SpiNNFrontEndCommon #364: duplicated code in reocrding.c

alan-stokes, 2018-10-25:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/c_common/front_end_common_lib/src/recording.c#L77-L85

and this method which is no longer being used and i fear no longer works with the new process. the private method needs to replace the public method. 

https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/c_common/front_end_common_lib/include/recording.h#L52-L59

rowleya, 2018-10-25:
These aren't quite the same, but I see what you mean.  I don't think there is an issue having both from the recording interface viewpoint though...

alan-stokes, 2018-10-26:
well having the recording flags seems only to be to support the .h call. The .c call seems to be the current way the recording api operates. So having that as a private method when it could replace the .h one and remove the recording flags..... id do it, less things to go wrong.

closed 2019-08-02 by andrewgait