SpiNNakerManchester/SpiNNFrontEndCommon #365: no clear mapping between python and c code recording regions

alan-stokes, 2018-10-25:
It ust to be that a variable had a given recording id and that it was handed down from python to the c code, thus avoiding mismatch between python and c. now the c has a hardcoded set of #defines which flag the index's, but there is no such fixment in the python, it only occurs due to a ordering of a list. this should instead be sent down to the c code, so that future changes do not cause this to crash.

Christian-B, 2018-10-25:
That is from long ago. I did not change that.  At least I don't think I did.

alan-stokes, 2018-10-25:
didnt say ya did. just that it does need fixing, and you've last been there. so ya would be best tailored to fix it. *grins*

rowleya, 2018-10-25:
Is the set of defines in FEC or should this be a sPyNNaker bug?

andrewgait, 2018-10-25:
The hardcoded set of defines is in sPyNNaker at https://github.com/SpiNNakerManchester/sPyNNaker/blob/master/neural_modelling/src/neuron/implementations/neuron_impl_standard.h so perhaps this is a sPyNNaker issue really.

alan-stokes, 2018-10-26:
probabily.

closed 2018-10-26 by alan-stokes