SpiNNakerManchester/SpiNNFrontEndCommon #381: test_app_finisher() inconsistant

Christian-B, 2019-01-03:
SpiNNFrontEndCommon\unittests\interface\interface_functions\test_front_end_common_app_finisher.py

Test sometimes passes sometimes not.
txrx.sdp_send_count  gives inconsitent counts

Has been happening for a while now.

alan-stokes, 2019-01-07:
given sdp is on UDP, isnt it kind of expected that the n messages might be inconsistent, as its based off network performance etc etc.

rowleya, 2019-01-07:
This uses mocking so it shouldn't matter that UDP would be used in theory.  I think the idea is that the state transistions over 3 requests, so it should take 3 requests to get to the finished state.  Is the failure always with the same value?

alan-stokes, 2019-01-07:
having looked at the test. i concur. lol. i just read "txrx.sdp_send_count gives inconsitent counts" and went, well yea. and assumed the test tested on machine. im just trying to run it (the computer is being very slow, lol).

alan-stokes, 2019-01-07:
found the issue i think. 

theres a race condition here. If your timer thread runs a bit slower than say the finisher, then the finisher could cycle through 2 iterations asking for finished state before your "txrx timer" switches state. As each time it finds a core which isnt in Finished, it sends the Prov and exit message. so that means there's a chance for it to go upwards in sdp messages by the race.

also, the test script is missing:
if __name__ == "__main__":
    test_app_finisher()

or 

if __name__ == "__main__":
    unittest.main()

as when i tried running it, it said there wasnt any tests........ till i put the code in.

rowleya, 2019-01-07:
I think this is a pytest script, so you don't normally need a name == "main" statement.  Normally instead run as a unit test, which should then run it correctly (provided your editor knows how to run pytest).

alan-stokes, 2019-01-07:
i tried, it said there was no tests

alan-stokes, 2019-01-07:
its worth noting this has a gotcha the other way, if the finisher missed timer 3. itll blow up due to array length overrun. maybe its worth just testing the end states that were reached? as i can see that its hard to know when to switch state, unless you encapsualte it into a class which changes when its asked for its state.

rowleya, 2019-01-07:
Fixed in #383

andrewgait, 2019-07-23:
This was fixed, so closing.

closed 2019-07-23 by andrewgait