SpiNNakerManchester/SpiNNFrontEndCommon #391: possible oddness

alan-stokes, 2019-02-06:
I have no idea if im just missing something. but if im not, then this code shouldn't be able to work:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/utilities/report_functions/memory_map_on_chip_report.py#L73

says its extracting the address in user1. But nothing I can find writes to user1. so that's going to make a mess of something. 

On a side note, it seems SpiNNMan is missing a call to user3 register. (ive added it in my branch, but was wondering why we omitted it, is there something im not aware of?)

rowleya, 2019-02-06:
I believe that is Sergio's code to read the memory map from the machine when running DSE on chip.  We don't really do DSE on chip any more, so the code is probably redundant.

No reason for the omission - probably just never came up!

alan-stokes, 2019-02-06:
phew. on the omission. I've added it in and am planning to use it for the router compressor with bitfield on chip. I was just chekcing that I hadnt misunderstood the code, as everything in spinnaker tools assumes a user3 exists, but we could have done some trickery that I've forgotten. So glad to know we've both forgotten if there is, and that we're convinced its safe. 

On user 1, phew, I was hoping you'd say that, I was worried I was going to trample over some other thing that's expecting some magic on user1 later on (I'm going to be trampling over enough stuff as it is). 

if its basically never going to be set, and thus never used, I should be safe to use it for my own wicked uses. wha ha haaa. 

 will close, and maybe delete the report in another pr at some point

closed 2019-02-06 by alan-stokes