SpiNNakerManchester/SpiNNFrontEndCommon #398: missing paused state in USES_SIMULATION_INTERFACE?

mpuzovic, 2019-03-01:
When running hello_world example from SpiNNakerGraphFrontEnd (https://github.com/SpiNNakerManchester/SpiNNakerGraphFrontEnd/tree/master/spinnaker_graph_front_end/examples/hello_world) 

python -m spinnaker_graph_front_end.examples.hello_world.hello_world

my understanding is that at the end the cores that are running hello_world binary will go to pause state. But, since vertices are defined to be of USES_SIMULATION_INTERFACE that as end state only has CPUState.Ready (https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/utilities/utility_objs/executable_type.py#L25) then in application_runner.py when waiting for end (https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/interface/interface_functions/application_runner.py#L136) it will raise an exception (https://github.com/SpiNNakerManchester/SpiNNMan/blob/master/spinnman/transceiver.py#L2028) because the cores are not in expected state.

If I add CPUState.PAUSED to USES_SIMULATION interface then the application shutdowns gracefully.

diff --git a/spinn_front_end_common/utilities/utility_objs/executable_type.py b/spinn_front_end_common/utilities/utility_objs/executable_type.py
index 2c344a7..6e56dc1 100644
--- a/spinn_front_end_common/utilities/utility_objs/executable_type.py
+++ b/spinn_front_end_common/utilities/utility_objs/executable_type.py
@@ -22,7 +22,7 @@ class ExecutableType(Enum):
     USES_SIMULATION_INTERFACE = (
         2,
         [CPUState.SYNC0, CPUState.SYNC1, CPUState.PAUSED, CPUState.READY],
-        [CPUState.READY],
+        [CPUState.READY, CPUState.PAUSED],
         True,
         "Calls simulation_run() and simulation_exit() / "
         "simulation_handle_pause_resume()")

andrewgait, 2019-03-01:
The GFE examples are currently broken because they haven't been updated correctly to match the C code style elsewhere.  The relevant branch is "simulation_interface_fix" on SpiNNakerGraphFrontEnd (see https://github.com/SpiNNakerManchester/SpiNNakerGraphFrontEnd/pull/100) which is waiting for someone to review it...

[Your method is also (I think) a valid way of dealing with the situation so perhaps having this patch on the python side in addition would not be a problem...]

alan-stokes, 2019-05-29:
isnt this now redundant thanks to andy gfe merge?

andrewgait, 2019-05-29:
Yes, I think so.

closed 2019-05-29 by andrewgait