SpiNNakerManchester/SpiNNFrontEndCommon #416: change n chip required to N boards required

Christian-B, 2019-04-29:
As spalloc and I assume the HBP portal which wraps spalloc can not actually do n chips it would be better to ask for n_boards.

On a very large number of chips

In my case 25914 chips spalloc actual returned less chips than asked for.

Would requires AbstractSpinnakerBase
1. to keep track of the user specified n_board
2. For backward compatibility allow users to still specify n_chips
3. Adapt the results of BasicPartitioner, PartitionAndPlacePartitioner and GraphMeasurer

Christian-B, 2019-04-29:
of course just because you ask for a specific number of boards you may get more.

For example asking for 480 boards gave me 507

Christian-B, 2019-04-29:
I notice that https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/interface/spinnaker.cfg
has config number_of_boards = None but I think this does NOTHING!

rowleya, 2019-04-29:
Yes, things come and go and that one has gone - that was the number of boards in a virtual machine, but was not needed...  I guess it could be useful to allow this to be used for a spalloc machine though

Christian-B, 2019-04-30:
Short term if anyone needs this
 a hack is to add (to SpallocAllocator.call)
        if n_chips < 0:
            n_boards = abs(n_chips)

And then do 
sim.setup(timestep=1.0, n_chips_required=-12)

Christian-B, 2019-06-05:
I think this should be part of the bigger project of getting the Application Vertexes to describe their machine needs and then get a machione based on that rather than partitioning on a fake machine to get machine needs

andrewgait, 2019-07-24:
I agree with @Christian-B, and as such this should probably be left for a future release.

alan-stokes, 2019-09-02:
I think this should be part of the bigger project of getting the Application Vertexes to describe their machine needs and then get a machione based on that rather than partitioning on a fake machine to get machine needs

this sounds more on the liens of the splitting api. But it still needs a machine, as it needs to do same chip as stuff for sdram edges etc. so unfortunately, the machine needs is a very complex thing that no 1 vertex can determine without a resource tracker to see what other verts have done so far. And the resource tracker requires a machine. .......

Christian-B, 2020-01-21:
n_boards_required has been implemented

closed 2020-01-21 by Christian-B