SpiNNakerManchester/SpiNNFrontEndCommon #427: SQLite stops multithreaded use

rowleya, 2019-05-15:
If the simulation is run in a different thread from the place where the data is extracted, the following error occurs:
```
File "C:\Users\zzalsar4\git\sPyNNaker8\p8_integration_tests\quick_test\test_auto_pause_and_resume_tests\test_run_forever_recorded\test_run_forever_recorded.py", line 46, in run_forever_recorded
    spikes = pop.get_data("spikes").segments[0].spiketrains
  File "C:\Users\zzalsar4\git\sPyNNaker8\spynnaker8\models\populations\population.py", line 259, in get_data
    return self._extract_neo_block(variables, None, clear, annotations)
  File "C:\Users\zzalsar4\git\sPyNNaker8\spynnaker8\models\recorder.py", line 55, in _extract_neo_block
    self._append_current_segment(block, variables, view_indexes, clear)
  File "C:\Users\zzalsar4\git\sPyNNaker8\spynnaker8\models\recorder.py", line 160, in _append_current_segment
    spikes=self._get_spikes(),
  File "C:\Users\zzalsar4\git\sPyNNaker\spynnaker\pyNN\models\recording_common.py", line 221, in _get_spikes
    sim.machine_time_step)
  File "C:\Users\zzalsar4\git\sPyNNaker\spynnaker\pyNN\models\neuron\abstract_population_vertex.py", line 504, in get_spikes
    placements, graph_mapper, self, machine_time_step)
  File "C:\Users\zzalsar4\git\sPyNNaker\spynnaker\pyNN\models\common\neuron_recorder.py", line 194, in get_spikes
    placement, region)
  File "C:\Users\zzalsar4\git\SpiNNFrontEndCommon\spinn_front_end_common\interface\buffer_management\buffer_manager.py", line 643, in get_data_by_placement
    placement.x, placement.y, placement.p, recording_region_id)
  File "C:\Users\zzalsar4\git\SpiNNFrontEndCommon\spinn_front_end_common\interface\buffer_management\storage_objects\buffered_receiving_data.py", line 231, in get_region_data
    return self._db.get_region_data(x, y, p, region)
  File "C:\Users\zzalsar4\git\SpiNNFrontEndCommon\spinn_front_end_common\interface\buffer_management\storage_objects\sqllite_database.py", line 142, in get_region_data
    data = self._read_contents(c, x, y, p, region)
sqlite3.ProgrammingError: SQLite objects created in a thread can only be used in that same thread.The object was created in thread id 14496 and this is thread id 15432
```

It might be possible to avoid this by checking the thread is the same when the data is extracted.

Christian-B, 2019-05-15:
why are you reading data from a different thread to where the database was opened?
Could this be a case of two separate runs sharing the same reports folder and therefor the same database file? If yes then that is complete MESS!

note I say opened because in many cases it is java which writes to the database and then python reopens the database.

And you wonder why we dont recommend multithreading in python. sqllite programmers feel the same

rowleya, 2019-05-15:
This was just something I was trying.  I had another way to do it, but this used to work, and now doesn't.  I don't expect it to be hard to support and it just removes another barrier to use.  Note that matplotlib does not work well in threads, so threading our code when using matplotlib live for example might be useful at some point.

dkfellows, 2019-06-15:
We would need to make the actual database handles be in thread-local storage and ensure that nothing holds a lock for very long. It's very fiddly (and I do something like it in the Java code, which definitely is multithreaded).

If there isn't a *pressing* need, I suggest not doing this for now...

dkfellows, 2023-09-27:
We simply aren't ever going to support this mode of usage.

closed 2023-09-27 by dkfellows

Christian-B, 2023-09-27:
Actually we do now support this model.

the data is extracted and save in near raw format.
Then the database is closed.

We then later use the NeoBufferDatabase see https://github.com/SpiNNakerManchester/sPyNNaker/blob/master/spynnaker/pyNN/utilities/neo_buffer_database.py#L43 to read the data.

This creates a new database handler so can easily be on a different thread or even a different Python run(if you hard code the path)