SpiNNakerManchester/SpiNNFrontEndCommon #500: optimisation for speed up

alan-stokes, 2019-10-14:
having starting writing this up. i ntoiced that we have some dangling stuff in the python code to adapt to if we were ever to use MC packets (which we removed long ago)  lines 226 to 233 in extra mon machine vertex. 

as well, these 4 keys are hard coded in the python. why don't we hardcore them in the c code and remove 16  bytes of data to load on every chip?

Christian-B, 2019-10-14:
No idea why I am being asked to clean up @alan-stokes code

alan-stokes, 2019-10-14:
booooo. there was me thinking id off loaded this. lol

Christian-B, 2021-03-22:
this appears to be a plan to change ExtraMonitorSupportMachineVertex method  _generate_data_speed_up_out_config

This is based on DataSpeedUpPacketGatherMachineVertex.TRAFFIC_TYPE which is hard coded to EdgeTrafficType.FIXED_ROUTE

Christian-B, 2021-03-22:
We we agree that DataSpeedUpPacketGatherMachineVertex will never use MULTICAST ?

If yes this makes sense and is an easy fix.

rowleya, 2021-03-23:
Happy with it if we never use it currently as it seems to work as it is.  I will note that SpiNNaker-2 won't have fixed-route packets, but will also likely be quite different in how data in and out is managed so lets simplify for now and fix if we have to later.

dkfellows, 2023-09-27:
We don't plan to do anything about this as on networks of any real size we're already saturating the routers connecting the SpiNNaker1M machine to the system(s) hosting the VMs behind all main public-facing access paths. Actually improving bandwidth would require moving larger UDP packets as that is now the principal bottleneck (and a hard one to resolve due to the design of SCAMP); everything else simply won't make much difference.

closed 2023-09-27 by dkfellows