SpiNNakerManchester/SpiNNFrontEndCommon #521: ds_pretend_database in FUBAR

alan-stokes, 2019-11-21:
this class located in fec->interface->ds->ds_pretend_database is not used by anything that i can see. But what adds insult to injury is that it cant be instantiated as its missing at least 2 abstract methods (clear_ds, ds_mark_as_system) (i will be adding a third shortly) which means its usage is already broken. 

can we remove or do we need to fix it?

alan-stokes, 2019-11-21:
whilst i'm at it. why have a abstract (ds_abstract_database) if only 1 thing inherits from it?

Christian-B, 2019-11-25:
Accidentally fixed straight to master:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/commit/4ca70f9251466e78cdb65b1d37487768ffcfb45e

Christian-B, 2019-11-25:
Update checked by @alan-stokes so closing issue.

closed 2019-11-25 by Christian-B