SpiNNakerManchester/SpiNNFrontEndCommon #524: Should we allow run timess that are not a multiple of timestep

Christian-B, 2019-11-27:
Script:
import spynnaker8 as sim
sim.setup(timestep=10.0)
print("get_current_time", sim.get_current_time())
pop_1 = sim.Population(1, sim.IF_curr_exp(), label="pop_1")
sim.run(25)
sim.run(25)
sim.run(25)
sim.run(25)
print("get_current_time", sim.get_current_time())

What do you expect the results to be?

Suggestion:
Throw an error

rowleya, 2019-11-27:
I believe that we have had this debate before... see https://github.com/SpiNNakerManchester/sPyNNaker/issues/149.

To summarise, the user preference was to warn and round.  In the case you provided, I would suggest rounding up.  Because the user is then warned, they shouldn't complain that the current time ends up being 120.

So, in summary, I expect the current time to be 120 (25 rounded up to the next multiple of 10 4 times).

Christian-B, 2019-11-27:
Based on https://github.com/SpiNNakerManchester/sPyNNaker/issues/149 we expect a warning and a round up.

Well OOPS! No warning and a round DOWN! 
So converting to a bug!

Christian-B, 2019-11-27:
I notice that 
sPyNNaker8/spynnaker8/spinnaker.py method _run_wait
does the round up
logs the run with the roundup.
NO warning

Then merrily throws the roundup away!

Christian-B, 2019-11-27:
While AbstractSpinnakerBase method _run
rounds DOWN!

Again no warning!

rowleya, 2019-11-28:
Fixed in master

alan-stokes, 2019-12-02:
just a note that may have been missed previously. what do you do if the end user script looks like this:

import spynnaker8 as sim
sim.setup(timestep=10.0)
print("get_current_time", sim.get_current_time())
pop_1 = sim.Population(1, sim.IF_curr_exp(), label="pop_1")
pop_2 = sim.Population(1, sim.SpikeSourceArray([0, 1, 25]))
sim.run(25)
pop_2.set_spike_times([26, 30, 40])
sim.run(25)

if you've rounded up, you have now lost the 26. and maybe even the 30. And i presume there's no warnings taken there. 

This also has a add on effect to the Ear, where the lowest common multiple effect.

rowleya, 2019-12-02:
It looks like it works but it doesn't warn.  The actual operation is this:
```
def _as_numpy_ticks(times, time_step):
    return numpy.ceil(
        numpy.floor(numpy.array(times) * 1000.0) / time_step).astype("int64")
```

Now, I'm not sure if this means it ceils or it floors, but it certainly doesn't warn!

Christian-B, 2019-12-02:
While Alan is correct that the above example would loose spikes the choice to round up has been taken at the PyNN level so is one so is one we have to live with.

Best we could do is add a Strict config that if one would raise an exception anywhere we do a user is an idiot

Christian-B, 2019-12-02:
This is defined by PyNN so no need to reopen!

closed 2019-12-02 by Christian-B

alan-stokes, 2019-12-02:
the thing i think im tying to point out. is that if your going to warn users instead of hitting them with a smoked kipper, then we need to warn them at every point where their actions have done something odd. regardless of if the spike times are rounded or not. its not being fired at the timestep the user expects. Heck, even if its rounded down, the SSA c code i think throws it away as its current timestep and the buffer time stamp are out in a stale way. 

Thus if your going for warnings, we need to check the spike times are not stale, and if any of them are, warn the users that those spikes are not going to be sent.