SpiNNakerManchester/SpiNNFrontEndCommon #526: Globals_API instead of passing config values all over the place.

Christian-B, 2019-11-28:
There are several values that remain constant from the end of sim.setup until the sim.end.
Yet these are often passed all over the place.

The idea is to define a globals_api which would provide these values.
All the alogirothm could then access these directly and cleanly.

Under the hood the implementation would probably be
globals_variables.get_simulator().foo

But to the Algorithm it would be simply
globals.foo

The values would include
The two key ones are
machine_time_step 
time_scale_factor

But it could include nearly all of the values defined in the cfg but best left to just any accessed anywhere else than ASB

It should then even be possible to separate ConfigHandler and AbstractSpinnakerBase
So instead of having to pass the whole Simulator we are just passing ConfigHandler 

The API should be defined at the PACMAN level with the config based implementation deferred to SpiNNFrontEndCommon.  
PACMAN test could have a stub implementation with fixed values.

Open Question is should/could this include values changed at each run like:
report_folder
run_time
ect

Probably not in the first pass!

alan-stokes, 2019-12-02:
well this will majorly conflict with the multiple timestep stuff. esp as that doesn't want a machine_time_step. 

The other thing you must consider is algorithms which are not python based. these cannot access the globals and yet may need those params. Thus the reason they are inputs to the work flow manager

Christian-B, 2019-12-02:
I am not worried about conflict with other branches. That should never be a reason to turn down a good idea. 

I am also not worried about the none python based one. For one we don;t have any and for two it would be easy enough to pass this in at that point.

So that leave just the discussion if this is a good diea not if there ay short trm issues to fix.

alan-stokes, 2019-12-02:
1. we dont, but other users could do. we've supported none python algorithms since.... well since we had algorithms. 

The other issue is that i currently cant see how youd bring them over, without making a none python algorithm into a python algorithm just to get inputs. which makes the whole idea of supporting none python algorithms pointless.  

If the idea is good or not....... your proposing making 2 apis to get data. instead of the 1 we currnetly have. Just coz the data is static once built. 

The logic flow is that most items are unmutable once built, so your going to fuck up the work flow manager majorly going down this route fully, as id argue that placements, routing,key alloc, tags etc should go down this route then, and thats going to make the workflow impossible. It will also make lives of developers of algorithms harder as they will not know which api to use for any given input when they run anyhow. 

The whole point of the workflow is that the algorithm developer does not need to give a toss of how to get the data, we'll hand it to them.  

so unless your proposing to augment the work flow manager to read the config as well as the inputs.....im against this idea.

So overall. i dislike the idea fully.

Christian-B, 2023-07-10:
done with Views

closed 2023-07-10 by Christian-B