SpiNNakerManchester/SpiNNFrontEndCommon #530: Strickt config Exception instead of warning

Christian-B, 2019-12-02:
Came out of https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/issues/524

We have a number of case where due to the PyNN specification we have to accept weird behaviour do a warn and run as best as we can.

For example:
run with no graph
run with a time not a simtime
selective recording for populations that dont support hat
pop.get with gather =False
compatible_output=False
Population not supporting units for a variable

this config would be off by default but would be set to True in debug mode

alan-stokes, 2019-12-02:
just to check something that sounded off. 

"Population not supporting units for a variable". units of a variable it does support recording of? if that's the case, its a error on the impl. if its a units of a variable it cant record, then that's a error of GIGO. So im not seeing how you "warn and run as best as we can." in any use case in this view. 

"selective recording for populations that dont support hat"
so i ack that ssa and ssp dont yet support that, and thus good idea, but my question is how hard would it be to add that stuff to their recorders? 

"run with a time not a simtime" where the heck is that supported?

Christian-B, 2019-12-02:
"Population not supporting units for a variable"
https://github.com/SpiNNakerManchester/sPyNNaker8/blob/7c575dc7d8b1fcd02bf2c57dff24683339c10c1f/spynnaker8/models/recorder.py#L92

"run with a time not a simtime" where the heck is that supported?
timstep oops fixed above.

alan-stokes, 2019-12-02:
that looks very wonky. given your basically saying " i know better than the neuron model" which is well wrong. 

esp as this exists:
https://github.com/SpiNNakerManchester/sPyNNaker/blob/master/spynnaker/pyNN/models/neuron/abstract_population_vertex.py#L835-L842

which shows that the model checks that the variable the units are being requested for are actually one that can be recorded. which given your access is in the recorder, can only be true you'd hope. 

this feels like a bodge to fix some unknown thing, likely some failure in a neuron impl. 

even worse your implying the only things that can be recorded are those in the _DEFAULT_UNITS, which even with some of our models is not true. SSA ONLY has spikes afterall.

Christian-B, 2019-12-02:
This issue is to warn where we MUST allow the users to input stuff we don't want to support but have to so do something which may be expected. So just do a warning.

The config would turn this behavior off.

If there is stuff we NEVER should accept than that is a seperate PR to remove the offending code altogether even with Strick = false

alan-stokes, 2019-12-02:
i agree that your pr has merit. just that where you've described this use case or units, i don't see it as a "must allow GIGO", as theres lots of parts here where either your making sweeping assumptions that dont hold, or where it just dont make sense. how do you respond when you say, ask a SSA for a units for membrane protential. your going to return what that means for a IfCurExp, but the SSA DOESNT HAVE a membrane protential, so its GIGO, and id rather slap the idiot with a wet kipper in this case. heck, if your looking at it fromt eh reocrder point, itll have rte'ed before reaching the neo object, as there ISNT any membrane protentail variable to read recorded data for.  

HECK! the only way a end user can get to units directly froma  pop/pop view/ assembly is via find_units. which says in the pynn docs:

find_units(variable)¶
Returns units of the specified variable or parameter, as a string. Works for all the recordable variables and neuron parameters of all standard models.
http://neuralensemble.org/docs/PyNN/reference/populations.html

it does not say "works for any random garbage". therefore making it return some garbage is not a correct move.

Christian-B, 2019-12-02:
@alan-stokes your comments about units is out of scope of this Issue.
Please create a new issue as yu are (I think correctly) suggestion a behavior change in all cases not just Strick=True

alan-stokes, 2019-12-02:
We have a number of case where due to the PyNN specification we have to accept weird behaviour do a warn and run as best as we can.

For example:
run with no graph
run with a time not a timestep
selective recording for populations that dont support hat
pop.get with gather =False
compatible_output=False
**Population not supporting units for a variable**

not out of scope when you listed it as one of the use cases you want to handle with a warning

Christian-B, 2019-12-02:
re:Units
It is already Handled with a warning.
This issue want to change that warning to an exception if Strick= True

@alan-stokes suggestion of never handling the exception and instead rethrowing it even when there is a usable _DEFAULT_UNITS may well be a valid suggestion (As it then says Vertex writer do your job properly!) but it is out of scope for this PR.

alan-stokes, 2019-12-02:
and what i'm trying to say. is there's no warning that's suitable, as a warning assumes you can recover, which if you don't have the variable, or if the variable has no units, both of these are valid smoked kippers, and thus not warnings at all. regardless of strict = False or True. thus ya might as well fix it properly and make it throw an exception the moment shit hits the fan there.

also before you go to far. Strick vs Strict? (I had to go google to get the correct spelling myself).

Christian-B, 2019-12-02:
Spelling fixed thanks.

Out of scope comment ignored

Christian-B, 2020-02-07:
https://docs.python.org/3.8/library/atexit.html

Christian-B, 2020-03-17:
fixed with https://github.com/SpiNNakerManchester/SpiNNUtils/pull/109

closed 2020-03-31 by dkfellows