SpiNNakerManchester/SpiNNFrontEndCommon #534: get_current_time after runforver

Christian-B, 2019-12-13:
if a run forever has happened and no reset has been called.

How should get_current_time act?

Currently it raises some kind of None exception

Assigen to @Christian-B with the desired behaviour

alan-stokes, 2019-12-16:
honestly. id say it should just return None and a log warning with something along of the lines of "When running forever, it is not possible to determine the current runtime". 

Id also argue that when end is called after a run forever, it shoudl also return the same message, as until we implement some way to ensuring all cores are at a given timestep when pausing in a none auto pause and resume manner, it is not possible.

Christian-B, 2022-03-29:
In https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/pull/929 the choice was made to return 0.0

closed 2022-03-29 by Christian-B