SpiNNakerManchester/SpiNNFrontEndCommon #541: Provenance data writes too many files

rowleya, 2020-02-03:
During the execution of a large job (about 200 boards) the tools "ran out of disk space"; when I checked there was plenty of space (over 200GB), so I guess that the writing of a lot of files to a single folder is probably the issue.

Options include
 - create subfolders for the provenance files
 - create a database of provenance files

I think I prefer the latter as it means that the data is easier to organise and query.

rowleya, 2020-02-03:
It may also be nice to be able to disable "graph" and "on machine" provenance gathering in this particular case as it isn't important, where PACMAN provenance is for statistical purposes; that said it is nice to know if the run actually worked!

alan-stokes, 2020-02-03:
silly question. but could the many users mean that it actually did run out of disk and that during some other run it cleared some reports before you got to check the disk space? It sounds odd to have a limit on how many files can be in a folder. 

i agree to the idea of being able to turn off separate prov items. A thought (instead of what would be a lot of cfgs, given @Christian-B 's inate grumble of too many dials) is to make a single thing in the cfg called prov algorithms. then you only need to switch out which ones you want? It likely means you can clear up a few cfgs such as "get prov items"

on the database, i dont mind that. but i dont think i want the iobufs in there..... mainly coz thats the one i read the most in debug, and adding a database api is another step in debugging.

rowleya, 2020-02-03:
This was my own run; I kept a close eye on the disk space during the run and it didn't dip below 200GB, so it must be some other limit causing the issue.

Any sort of switch is useful, so yes a global provenance plus individual ones (like reports in general) is useful.

IOBUF outside of a database is also fine; I agree that this is mainly used for debugging in any case, so extra steps would be more painful to use.  That said, having the *option* of using a database could be nice for those cases with the same issue (so many cores that it fills up a folder).  I think we can leave that case until we cross it though.

alan-stokes, 2020-02-03:
i wasnt thinking like reports. which has something akin to 20 separate cfg dials. I was thinking more like machine_graph_to_machine_algorithms, machine_graph_to_virtual_machine_algorithms, loading_algorithms. As these are a accum of algorithms, and each prov item you mentioned has its own algorithm that ASB puts in based off the cfg param "write_provenance_data". If you had something akin to machine_graph_to_machine_algorithms. youd get rid of a lot of if's in ASB and would give you your single dial.

aka 
provenance_algorithms=PlacementsProvenanceGatherer,GraphProvenanceGatherer,RouterProvenanceGatherer,ProvenanceXMLWriter,ProvenanceJSONWriter,ProfileDataGatherer

alan-stokes, 2020-02-03:
weirdly it would simply ASB which would make @Christian-B more happy. I retract my idea accordingly ;-)

rowleya, 2020-02-03:
Yes, happy with that idea too!

Christian-B, 2020-02-10:
I would be against adding provenance_algorithms ....  
***_algorithms should only be used to provide alternative code for steps.

Instead I would change provenance writing to reports.
Add dials to the [reports] section of the cfg.
I have nothing against independent on/off dials in the cfg.  it is dials that effect each others that are dangerous. 

These can the also be turned on and off with the mode production/debug

That said if there is an issue where provenance/ reports/ iobuff ect can write to many files that also needs fixing.

rowleya, 2020-03-30:
This has been fixed in #546

closed 2020-03-30 by rowleya