SpiNNakerManchester/SpiNNFrontEndCommon #552: why do we still have app data folder?

alan-stokes, 2020-02-25:
why do we still have app data folder? if the only thing going into it these days is a input_output_database that we have control over. and that we report the path to for end users?????

Christian-B, 2020-02-25:
@dkfellows suggests checking reloading

Christian-B, 2020-02-25:
I take this as permission from @alan-stokes to nuke it if I can

dkfellows, 2020-02-25:
I suspect it isn't possible yet. It might be possible to postpone creating it until we need to put something inside.

alan-stokes, 2020-02-25:
define reloading exactly. given isnt reloading going via your database? for the DSE component.

alan-stokes, 2020-02-25:
the main reason this came up is a master student https://github.com/jofas here in EPCC asking where the data was. and i got a bit confused till i remembered its all in the ds.sql. at which point this question fell out of the logic.

dkfellows, 2020-02-26:
I _think_ we ought to be thinking of the app data folder as being on Death Row at this point. Fortunately it doesn't smoke so we don't need to mess around with that when we line it up against the wall with a blindfold on…

Christian-B, 2020-02-27:
It is used by https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/interface/interface_functions/dsg_region_reloader.py
But only as a parent for a temp directory it removes at the end of its work.

Christian-B, 2020-02-27:
it is used by https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/interface/interface_functions/database_interface.py to put the db.sql database in

Christian-B, 2020-02-27:
It is used by https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/interface/interface_functions/write_memory_io_data.py to write data

Christian-B, 2020-02-27:
So the open question is do we ever use WriteMemoryIOData?

If NO shall we delete that algorithm?

If Yes this issue is a no can do!

Christian-B, 2020-02-28:
Discussed with @rowleya 
DSGRegionReloader put temp in reports in case it crashes so easy to find
DatabaseInterface put in reports/runX as that is where user would look

WriteMemoryIOData is a dead end idea put in for PhD students who are not using it.
    Put it in reports/run until WriteMemoryIOData breaks and gets nuked.

closed 2020-03-12 by alan-stokes