SpiNNakerManchester/SpiNNFrontEndCommon #564: Possible breakage in SystemMulticastRoutingGenerator

dkfellows, 2020-03-06:
At the end of `SystemMulticastRoutingGenerator._add_routing_entries` we see this:

https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/f5f1741a111b804a921af59b0537c1a471dc5787/spinn_front_end_common/interface/interface_functions/system_multicast_routing_generator.py#L180

That line is weird because it appears to be doing an increment that cannot have any effect. Should it be indented one level further so that it goes inside the loop immediately above it? That'd make sort of sense from a general-algorithmic-pattern sense.

dkfellows, 2020-03-06:
If it isn't in the loop, then it's still junk code. I think…

rowleya, 2020-03-06:
Looks like junk to me...

closed 2021-03-23 by rowleya