SpiNNakerManchester/SpiNNFrontEndCommon #578: Allow gathering of provenance without writing it

rowleya, 2020-03-30:
There is currently only one provenance option, which is to "write_provenance_data".  It would be useful if there was a "read_provenance_data" which would get and evaluate the provenance data (i.e. print warnings) without then storing it.  This could be useful when there is a lot of data and storage would be prohibitive.

alan-stokes, 2020-04-17:
fair enough.

alan-stokes, 2020-04-17:
does this still count given the prov via database fixes?

alan-stokes, 2020-05-05:
isnt this done now?

rowleya, 2020-05-05:
No, I think it is still written to the database.  It should be possible to store it in memory and then use it in the top-level script.

alan-stokes, 2020-05-07:
ok. so this is coz we use the write_prov config to control both the addition of the algoirtmhs, and the writing to whatever writer we set up to use. 

so a extra cfg param for reading them would solve that. 

i note that lines 2640 - 2657 and 1888 - 1909 are near clones of each other. i admit 1 tries to read the power mon prov. but if it didnt exist it would still work. so id assume it would be possible to strip them into 1 method for saving some lines. 

anyhow. might do this when got a bit of slack

alan-stokes, 2020-05-14:
closing this now that the fix is in. no idea why github didn't automatically do so. 
reopen if someone disagrees

closed 2020-05-14 by alan-stokes