SpiNNakerManchester/SpiNNFrontEndCommon #593: Encountered an issue within the live_packet_gather.c file (flush_events seems to corrupt the circular_buffer 'with_payload_buffer')

jofas, 2020-05-04:
Hi everybody,

currently I am trying to turn the Conways example from the [`SpiNNakerGraphFrontEnd`](https://github.com/SpiNNakerManchester/SpiNNakerGraphFrontEnd/tree/master/spinnaker_graph_front_end/examples/Conways/partitioned_example_b_no_vis_buffer) into a version that streams the state of each cell for each time-step back to the host (you can find the code [here](https://github.com/jofas/master_thesis/tree/lpg_debug/learn_spinn)). For this I am using a `LivePacketGatherMachineVertex` instance. 

After debugging the program with @alan-stokes, we found out that the `flush_events` function somehow corrupts the `circular_buffer` '`with_payload_buffer`'. The following version of the `incoming_event_payload_callback` function:

```C
static void incoming_event_payload_callback(uint key, uint payload) {
    log_info("Received key %x, payload %x", key, payload);
    circular_buffer_print_buffer(with_payload_buffer);
    log_info("size of buffer: %d", circular_buffer_size(with_payload_buffer));

    if (circular_buffer_add(with_payload_buffer, key)) {

        log_info("buffer before adding: ");
        circular_buffer_print_buffer(with_payload_buffer);
        log_info("size of buffer: %d", circular_buffer_size(with_payload_buffer));

        circular_buffer_add(with_payload_buffer, payload);

        log_info("buffer after adding: ");
        circular_buffer_print_buffer(with_payload_buffer);
        log_info("size of buffer: %d", circular_buffer_size(with_payload_buffer));

        if (!processing_events) {
            processing_events = true;
            spin1_trigger_user_event(0, 0);
        }
    } else {
        provenance_data.number_of_overflows_with_payload++;

        log_info("overflow for key: %x with payload: %x", key, payload);
        circular_buffer_print_buffer(with_payload_buffer);
        log_info("size of buffer: %d", circular_buffer_size(with_payload_buffer));
        log_info("not full: %d", _circular_buffer_not_full(with_payload_buffer));
        log_info("not empty: %d", _circular_buffer_not_empty(with_payload_buffer));
    }
}
```

produces as log for a single call:

```
[INFO] (live_packet_gather.c: 319): Received key b, payload 0
[]
[INFO] (live_packet_gather.c: 321): size of buffer: 0
[INFO] (live_packet_gather.c: 342): overflow for key: b with payload: 0
[]
[INFO] (live_packet_gather.c: 344): size of buffer: 0
[INFO] (live_packet_gather.c: 345): not full: 0
[INFO] (live_packet_gather.c: 346): not empty: 0
```

This happens after ```flush_events``` is triggered the first time with a `buffer_index > 0`. Before the lgp works fine.
Once we commented out the content of the ```flush_events``` function, we didn't enter the `else` clause when we ran a simulation.
The `circular_buffer` somehow thinks it is full even though it is empty.

Also, using [this old version](https://raw.githubusercontent.com/SpiNNakerManchester/SpiNNFrontEndCommon/6bf9b6504b8d4bec62bae09bf745116c2f024a23/c_common/models/live_packet_gather/src/live_packet_gather.c) of the lpg worked once @alan-stokes made it use the newer APIs.

I hope this can help the team diagnose this and fix as quickly as possible.

Kind regards,
Jonas

dkfellows, 2020-05-21:
6baae1fb2ab00d1d460c35fc4ae1e4692b78bd08 would have been a less odd commit to check (it's the merge commit very slightly ahead of the one you chose).

dkfellows, 2020-05-21:
And the “problem” is that the size of the buffer is zero.

dkfellows, 2020-05-21:
(It sounds like something has either failed to initialise the buffer correctly, or has somehow set its size to a bogus value.)

jofas, 2020-05-21:
> It would be helpful to see the result of `circular_buffer_real_size(with_payload_buffer)`, as that really shouldn't be zero.

the real size of the buffer is 511 until the second time `flush_events` is executed. Afterwards it is -442503144.

dkfellows, 2020-05-21:
OK, that's definitely memory corruption.

dkfellows, 2020-05-21:
After quite a bit of investigation, the problems appear to be stemming from the fact that the version of the `circular_buffer.h` and the initialisation code (in `libspinn_common.a`) aren't matching. Get a complete build of everything with matching headers and init code and your code (I've tested version ID: 21a8df51b9c66a8716010a9184810359e0b4f50e) appears to function correctly.

dkfellows, 2020-05-21:
Re why things not matching matters, that's because the circular buffers were changed (in https://github.com/SpiNNakerManchester/spinn_common/pull/30) to be more efficiently managed in terms of memory, but this had the side effect of making them use a different memory layout. C, being the language it is, provides _no_ protection against getting in a mess if this happens.

dkfellows, 2020-05-21:
Spoke too soon. 5&times;5 crumbles.

dkfellows, 2020-05-21:
The bug is here:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/7dbb7ee362414098823a677e2843ef8efcb96906/c_common/models/live_packet_gather/src/live_packet_gather.c#L158
The number of bytes to clear is being calculated to be twice as large as it should be when payloads are present. The resulting buffer overrun has many effects, but one of the main ones is to set `simulation_ticks` to zero, which causes the LPG to stop a little bit later in `timer_callback()`.

I _think_ we can just delete the clearing code; we are going to overwrite those bytes before we send them out anyway.

jofas, 2020-05-21:
Yes, thank you so much.  Commenting out the part resetting the content did the trick for me. I tested running 5x5 and 7x7 conway's cells.

Kind regards,
Jonas

dkfellows, 2020-05-21:
Issue will be closed when we get a fix in. Probably tomorrow (Friday).

PR #604 (2020-05-26): Prevent a buffer overrun in the LPG when using packets with payloads
The overrun was because of a miscalculation of the number of words to clear after sending a message, yet it turns out that you don't need to clear the memory at all.

Also factor out all the actual writes (not of just the unaligned words) to helper inline functions to make the code easier to debug.

Fixes #593
commit e0e33070 (2020-05-26): Prevent a buffer overrun in the LPG when using packets with payloads

The overrun was because of a miscalculation of the number of words to
clear after sending a message, yet it turns out that you don't need to
clear the memory at all.

Also factor out all the actual writes (not of just the unaligned words)
to helper inline functions to make the code easier to debug.

Fixes #593
closed 2020-05-26 by dkfellows