SpiNNakerManchester/SpiNNFrontEndCommon #623: ReverseIPTagMulticastSourceMachineVertex has different  meanings for send_buffer_time is None

Christian-B, 2020-06-29:
The class ReverseIPTagMulticastSourceMachineVertex has different  meanings for send_buffer_time is None
And this effects the sdram cost for recording

in method _recording_sdram_per_timestep(...
the assumption is send_buffer_times is None means  recording live data

in the init and empty  send_buffer_times is converted to None.  
However an empty  send_buffer_times param does not indicate live data.

Caught in spy8 integration test  TestSpikeSourceArray.test_recording_with_empty_lists
IF ReverseIpTagMultiCastSource.create_machine_vertex(....
has an extra
        if resources_required:
            assert (resources_required == machine_vertex.resources_required)

rowleya, 2020-06-30:
I assume you mean what happens if someone creates a SpikeSourceArray with no spikes and then asks to record it?  I can see why that would go wrong...

alan-stokes, 2020-06-30:
well...........what happens when you say use it for spike injection?

rowleya, 2020-06-30:
From memory alone I think you can record live spike input; it reserves "some space" for that and then fails to record if you go over that space.  I believe that this is user controlled, but again only from memory.

rowleya, 2020-06-30:
The issue here though is that an empty SSA with recording should require very little SDRAM, but it probably reserves enough for live spikes in that case as it doesn't see the empty SSA as being predetermined and fixed.

alan-stokes, 2020-06-30:
if it rserved enough for live spikes. it should still agree with the app vertex resources. the issue is that it doesn't. 

thus ReverseIPTagMulticastSourceMachineVertex's resources_required
and 
ReverseIpTagMultiCastSource's get_resources_used_by_atoms

dont match results, Which is bad bad

alan-stokes, 2020-06-30:
and funnily. ReverseIPTagMulticastSourceMachineVertex's  get_sdram_usage and corrasponding dtcm and cpu calls are static method. So why the app vertex makes its own logic...............

rowleya, 2020-06-30:
Yes odd - the app vertices are supposed to use the machine vertex in that case I believe

Christian-B, 2020-06-30:
To be clear the issue is caused not by the sdram cost function which as @rowleya points out is the same function but by how send_buffer_times is handled. IE what is passed into that method

In the init of ReverseIPTagMulticastSourceMachineVertex there is code:
        if n_buffer_times == 0:
            self._send_buffer_times = None

However that causes there to be two different meanings to self._send_buffer_times == None
1. n_buffer_times == 0
2.  Recording live data

The method _recording_sdram_per_timestep
gives different results for an array of empty arrays then it does for None!

alan-stokes, 2020-06-30:
to be clear, the _recording_sdram_per_timestep is in ReverseIPTagMulticastSourceMachineVertex. so if you ensured the app vertex called the machine vertex sdram calcs. wouldnt it be at least consistent?

Christian-B, 2020-06-30:
The App vertex for the case the case that fails calls
ReverseIPTagMulticastSourceMachineVertex.get_sdram_usage(
                send_buffer_times=[[],[]] ....

The MachineVertex calls
self.get_sdram_usage(
               send_buffer_times=None....

These two calls to the same function give different results

Christian-B, 2020-06-30:
> to be clear, the _recording_sdram_per_timestep is in ReverseIPTagMulticastSourceMachineVertex. so if you ensured the app vertex called the machine vertex sdram calcs. wouldnt it be at least consistent?

Not if called with different send_buffer_times parameter!

alan-stokes, 2020-06-30:
so 2 issues to fix. maybe make it 2 params. send buffer times and is running live recording

Christian-B, 2020-06-30:
I dont understand by the send_buffer_times=[[],[] in the int needs to be recorded as None

rowleya, 2020-06-30:
OK, so sounds like 2 problems:
 - An empty array is replaced by None.  Easy enough to fix I would think...
 - The application vertex might be split into several machine vertices, which might split up the send_buffer_times array.   I am guessing this would be fixed by breaking up the send_buffer_times using the given vertex_slice at partitioning time?

Christian-B, 2020-06-30:
> * The application vertex might be split into several machine vertices, which might split up the send_buffer_times array.   I am guessing this would be fixed by breaking up the send_buffer_times using the given vertex_slice at partitioning time?

That is already done correctly by the application vertex both when it does the costing and when it does the create vertex.
The issue is that in this weird test the split resulted in an array of empty arrays and caused the None

rowleya, 2020-06-30:
Ah OK.  That is a bit odd then!  So all is working except an array of empty arrays...  Well it is an odd use case for sure!

andrewgait, 2020-07-07:
This can be made to work if the array of empty arrays on a particular slice is changed to None.  Going in the other direction and trying to deal with sending empty arrays rather than None leads to more problems than it's worth dealing with for this edge case (specifically, sending empty arrays into particular functions rather than None messes up where the buffer manager  is expecting to read from next when extracting).

[This solution is in https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/pull/625]

Christian-B, 2020-07-07:
Ah, Ok

It is a rare edge case and I except the easier to do both wrong

closed 2020-07-09 by Christian-B