SpiNNakerManchester/SpiNNFrontEndCommon #635: Incorrect use of synchronisation barrier

rowleya, 2020-07-20:
The following code starts an application, but then uses a previous state of the synchronisation barrier.  The sync state is actually a property of sark, which means it is reset on each core when a new binary is loaded.  Thus this code should not use the synchronisation state from a previous run.

https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/115a6d5d811413d4aeedf556382549b391e0ef52/spinn_front_end_common/utilities/system_control_logic.py#L60

alan-stokes, 2020-07-20:
"reset on each core when a new binary is loaded"

so thats interesting, but also means we have issues if tehre is ever any system binary that uses sync at all. 

i persumed the signal updated the sync state regardless of binary............that being said, bitfields were the first second binary that had a sync requirement.....

rowleya, 2020-07-20:
You really need to keep track per execution per core; I think we have done this in an application before, because the whole application needs to be synchronised and isn't stopped in general, and also can be re-run, so there are multiple synchronisation points.  System binaries are different, but if you are only going to sync once, you don't need to keep track of the state.

rowleya, 2020-07-20:
Basically, there is an issue only if you have different long running binaries that need to be synchronised more than once.  Thus I think you can remove the synchronisation count from this method as it is designed for applications that run and stop within this call I think.

alan-stokes, 2020-07-20:
im happy to accept your point. just we do run a risk if we do have binaries that use sync and last longer like the extra mons etc.

rowleya, 2020-07-20:
Well so far we haven't seen that in practice (extra monitor doesn't sync I don't think)

Christian-B, 2021-03-22:
has this been fixed?

rowleya, 2021-03-23:
Looks like it!

closed 2021-03-23 by rowleya