SpiNNakerManchester/SpiNNFrontEndCommon #658: When self-partitioning is implemented, review has_key

rowleya, 2020-09-07:
Many of the vertices have "has_key" as a parameter.  This is there in case edges get removed at a later stage to the extent that the vertex is in the graph but has no outgoing edges.  With self-partitioning, this should never happen.  This issue is here to remind us to look at this at some future stage and see if has_key can be removed.

Christian-B, 2020-09-14:
Even with self partitioning vertexes are partitioned first and then edges so you can still end up with vertices with no edge,

rowleya, 2020-09-14:
Yes, it is possible, but then an application vertex might be able to work out in advance that a machine edge won't have any other edges coming out of it and so not create the machine vertex.  Removing has_key isn't a requirement, just desirable, so if this doesn't happen in the first version this is fine.

rowleya, 2020-09-14:
(Or if it never happens, that is also fine)

Christian-B, 2020-09-14:
Working out if a Machine Edge is needed is very tricky until both source and target vertex have been partitioned.

But if the adding of delays is only done after all vertexes are partitioned this may still be relevant.

alan-stokes, 2020-09-15:
well i raise a silly question. 
i have a population which has 100 neurons and i split by say 10 atoms. 

it gets fed from a SSA and i record v from the pop. if tehres no edge coming out your going to remove my pop, but then i cant record v on it at all?

but you still dont want that pop to fire either......

rowleya, 2020-09-15:
True enough - there are definitely cases where you want a population but don't want it to fire.  You might even want to record spikes on it even though they don't go anywhere!