SpiNNakerManchester/SpiNNFrontEndCommon #677: GraphProvenanceGatherer why application level

Christian-B, 2020-10-14:
https://github.com/SpiNNakerManchester/SpiNNFrontEndCommon/blob/master/spinn_front_end_common/interface/interface_functions/graph_provenance_gatherer.py

1.        if application_graph is not None:
 there is always at least an empty app graph so this check is wrong

2. When there is no application level AbstractProvidesLocalProvenanceData why check all application items.
2a. How could there be application "Local" anyway?

alan-stokes, 2020-10-14:
discussed wuth @Christian-B and fix in different place

Christian-B, 2020-10-14:
Point 1 yes different fix. Point 2 why use app level at all still open

alan-stokes, 2020-10-14:
point 2. 
1. its only machine things that use it. 
2. im perfectly happy to remove

        if application_graph is not None:
            progress = ProgressBar(
                application_graph.n_vertices +
                application_graph.n_outgoing_edge_partitions,
                "Getting provenance data from application graph")
            for vertex in progress.over(application_graph.vertices, False):
                if isinstance(vertex, AbstractProvidesLocalProvenanceData):
                    prov_items.extend(vertex.get_local_provenance_data())
            for partition in progress.over(
                    application_graph.outgoing_edge_partitions):
                for edge in partition.edges:
                    if isinstance(edge, AbstractProvidesLocalProvenanceData):
                        prov_items.extend(edge.get_local_provenance_data()) 

from the GraphProvenanceGatherer

Christian-B, 2021-03-23:
AbstractProvidesLocalProvenanceData are DataSpeedUpPacketGatherMachineVertex and ProjectionApplicationEdge..

So a MachineVertex and an Application Edge currently which so both appliction and machine  / Edges and vertex.

A minor speedup could be gained at tthe lost of checking if anything else implemented it we dont know about. And it would be hard to cap the Abstract

closed 2021-03-23 by Christian-B