SpiNNakerManchester/SpiNNFrontEndCommon #715: DatabaseWriter BROKEN needs fixing or nuking

Christian-B, 2020-11-30:
spinn_front_end_common/utilities/database/database_writer.py is broken and apparently unused.

1. Who is using this where and how?

2. It has fallen well behind so needs fixing IF it is still used.

For example
for vertex in application_graph.vertices:
      isinstance(vertex, AbstractRecordable)

Yet all superclasses of AbstractRecordable are MachineVertex

rowleya, 2020-12-01:
The issue here is that the test that tests this is a bit broken; it sets create_database to True, so this bit of code will not be required to make it work:
https://github.com/SpiNNakerManchester/sPyNNaker8/blob/master/p8_integration_tests/quick_test/test_external_devices/spynnaker.cfg

I suspect this config file should be deleted and the tests run and they should then fail.

rowleya, 2020-12-01:
On checking again, the test for AbstractRecordable is mostly irrelevant.  I don't think that database value gets used, so that could be removed.  The check that matters is AbstractSupportsDatabaseInjection.  This appears to be fully in operation, so removing the config file above is probably a good idea.

Christian-B, 2020-12-01:
On Checking the  recorded value which AbstractRecordable was attempting to set was never read back out of the database

closed 2021-02-04 by alan-stokes