SpiNNakerManchester/SpiNNFrontEndCommon #717: MemoryPreAllocatedResources could be init in ASB

dkfellows, 2020-12-01:
If we made ASB initialise the preallocated resources with an empty `pacman.model.resources.PreAllocatedResourceContainer`, algorithms that manipulate it wouldn't ever have to worry about the value being `None`.

Christian-B, 2021-03-23:
ASB does initialise the preallocated resources with an empty pacman.model.resources.PreAllocatedResourceContainer.

In fact it does it twice. Once in get Machine and once in do mapping.

Christian-B, 2021-03-23:
The resource tracker has it as the second optional argument. So changing it would require an order swap which is an API change so not doing that for a minor cleanup.

Christian-B, 2021-03-23:
How the PreAllocatedResourceContainer is extended is Alanish but works.

Each time a new PreAllocatedResourceContainer is created and then the existing one is dumped in.

The gain is minor for an APi change so will not ifx.

closed 2021-03-23 by Christian-B