SpiNNakerManchester/SpiNNFrontEndCommon #734: Comprressor algorithm renaming

Christian-B, 2021-01-14:
We have currently 9 compression algorithms definded in xml
1. MundyOnChipRouterCompression
2. UnorderedOnChipRouterCompression
3. PairOnChipRouterCompression
4. MachineBitFieldUnorderedRouterCompressor
5. MachineBitFieldPairRouterCompressor

6. MundyRouterCompressor
7. UnorderedCompressor
8. PairCompressor
9. HostBasedBitFieldRouterCompressor

1 to 5 on chip 6 to 9 on host

What they do is
1. Mundy Algorithm Mundy aplx
2. Mundy Algorithm partially clean up
3. Pair algorithm
4. Mundy algorithm with bitfields
5. Pair algorithm with bitfields

6. Mundy Algorithm derigged
7. Pair algorithm but assuming unordered result
8. Pair algorithm 
9. Mundy algorithm and bitfileds

Issue is 2 and 7 have the same confusing name bit do very different things.

Suggested renaming
1. MundyOnChipRouterCompression
2. OrderedCoveringOnChipRouterCompression
3. PairOnChipRouterCompression
4. MachineBitFieldOrderedCoveringCompressor
5. MachineBitFieldPairRouterCompressor

6. OrderedCoveringRouterCompressor
7. PairUnorderedCompressor
8. PairCompressor
9. HostBitFieldOrderedCoveringCompressor

Plus add if easy
10.HostBitFieldPairRouterCompressor

alan-stokes, 2021-01-22:
does this mean this can be closed now?

Christian-B, 2021-01-22:
yes

closed 2021-01-22 by Christian-B